mantidproject/mantid #15353: There is no documentation page for the fit function Bk2BkExpConvPV

FedeMPouzols, 2016-02-17:
I can't find the RST page for this function: https://github.com/mantidproject/mantid/blob/master/Framework/CurveFitting/src/Functions/Bk2BkExpConvPV.cpp
But there is this old wiki page for it: http://www.mantidproject.org/BackToBackExponentialPV

There is an RST page for `NeutronBk2BkExpConvPVoigt`, which should not be confused with `Bk2BkExpConvPV`.

FedeMPouzols, 2016-02-17:
@Anders-Markvardsen : I think this is the case probably for more fit functions. I also came across this wiki page for example: http://www.mantidproject.org/CreateBackToBackParameters which I'm not sure is easy to find or linked from anywhere.

Anders-Markvardsen, 2016-02-19:
@FedeMPouzols There is a link to http://www.mantidproject.org/CreateBackToBackParameters from http://docs.mantidproject.org/nightly/fitfunctions/BackToBackExponential.html

PR #28331 (2020-09-02): Add Bk2BkExpConvPV documentation
**Description of work.**
This PR adds documentation for Bk2BkExpConvPV 

**To test:**
Build the docs
Check it makes sense

Refs #16480
Fixes #15353

*This does not require release notes* because **it is documentation**

---

#### Reviewer ####

Please comment on the following ([full description](http://developer.mantidproject.org/ReviewingAPullRequest.html)):

##### Code Review #####

- Is the code of an acceptable quality?
- Does the code conform to the [coding standards](http://developer.mantidproject.org/Standards/)?
- Are the unit tests small and test the c [... cut]
closed 2020-09-02 by martyngigg