mantidproject/mantid #15371: Refactor/split the EnggDiff GUI: calib, focus, pre-proc/event-mode

FedeMPouzols, 2016-02-18:
Now that it's been growing for some time and before it grows with new tabs it's time to split the calibration, focusing, and pre-processing, event mode stuff into separate presenters and views (the UIs are already split). Do this hopefully before we add new tabs for "single peak fitting", "multi peak fitting" and other areas.

FedeMPouzols, 2016-05-11:
This couldn't be done before adding the single peak fitting tab. Now that we have it and the multi-peak fitting tab is arriving soon, this becomes a very high priority.

josephframsay, 2018-04-10:
First, have a look at [this document](https://github.com/mantidproject/mantid/files/1894808/refinement-of-profile-parameters-with-polaris-data.pdf). Though it's written for Polaris, all the parameters except the ones in section 2.3 are relevant to ENGINX (and probably IMAT, if you get that far).

josephframsay, 2018-04-10:
Calibration Tab
---------

The purpose of the calibration tab is to find the diffractometer constants (ENGINX appear to not really care about DIFA - it's just set to zero throughout the GUI).

As input, it takes either:
1. A vanadium run number and a calibration sample (also sometimes refered to as ceria) run number
2. A `.prm` file (this willl usually have been calculated at some earlier time in the Calibration tab)

The workflow in case 1 is:
1. Calculate two vanadium workspaces (note, in `EnggDiffractionPresenter` you'll see references to `loadOrCalcVanadiumWorkspaces`, which has a look to see if these have already been calculated to spare itself the job. In practice I'm not convinced that these are saved anywhere - certainly it appears that they are recalculated every time #17424). The two vanadium workspaces we're calculating are the `VanIntegrationWorkspace` and `VanCurvesWorkspace` properties of [EnggCalibrate](http://docs.mantidproject.org/nightly/algorithms/EnggCalibrate-v1.html). These are both produced by [EnggVanadiumCorrections](http://docs.mantidproject.org/nightly/algorithms/EnggVanadiumCorrections-v1.html)
2. Pass both the above workspaces to [EnggCalibrate](http://docs.mantidproject.org/nightly/algorithms/EnggCalibrate-v1.html), which gives us our diffractometer constants as output
3. Output a calibration file (`.prm`) to the general calibration directory (if you're on Windows, `C:\EnginX_Mantid\Calibration`) and the user calibration directory (`C:\EnginX_Mantid\User\<rb_number>\Calibration`). These is currently done by a script in `EnggUtils`, which just replaces the diffractometer constants in a template `.prm` file with the calculated ones. Write out an individual file (in both locations) for every bank we ran a calibration on, and one for all banks together if we didn't run a cropped calibration
4. The ED GUI holds a `GSASCalibrationParms` object (accessed through the `IEnggDiffractionCalibration` interface) - update this to hold the calculated diffractometer constants, as well as the ID of the bank we ran calibration on

The workflow in case 2 is, read the diffractometer constants from the `.prm` file and follow step 4.

josephframsay, 2018-04-10:
Focus Tab
---------

The focus tab focuses a run

It takes a single run as input. It is sort of implied that this can be a run number or range of run numbers, but in practice this isn't the case - you have to go into the data archive and find the nexus file. It's probably worth speaking to scientists about how they actually want this to behave

The workflow is as follows:

1. Again, calculate your vanadium workspaces with EnggVanadiumCorrections
2. Pass these, the input run, and the bank IDs to focus to [EnggFocus](http://docs.mantidproject.org/nightly/algorithms/EnggFocus-v1.html) (we're looking for one focused run per bank per run)
3. Save the output of EnggFocus to the user focus directory and the general focus directory. There are a number of different formats used
  - `.dat` corresponds to SaveFocusedXYE
  - `.his` corresponds to SaveOpenGENIE
  - `.gss` corresponds to SaveGSS
  - `.nxs` corresponds to SaveNexus

- **Focus Cropped** - this means selecting only certain spectrum numbers for focusing, and corresponds to the SpectrumNumbers property of EnggFocus
- **Focus Texture** - similar to **Focus Cropped**, but using a grouping file instead of specifying spectrum numbers explicitly

Things to watch out for:

- Again, bank IDs - what do we do about cropped and texture focusing?

josephframsay, 2018-04-10:
Pre-processing tab
-------------------

As far as I can tell, development didn't get very far on this tab. It appears to rebin its input workspace and do nothing with the result - have a chat with scientists about what the point of this actually is

josephframsay, 2018-04-10:
Fitting tab
--------

This is already in its own MVP pattern, but there are still things to be done to it. It would be nice to have the asynchronous running of EnggFitPeaks in the model, not the presenter, and the EnggDiffMultiRunFittingWidget could do with being brought into it, to replace the current **Preview** tab (bear in mind that EnggDiffMultiRunFittingWidget doesn't have a Peak Picker)

josephframsay, 2018-04-10:
GSAS-II Refinement tab
---------------------

~~is perfect and you never need to change it~~

There's clearing up to here as well:

- Instrument parameter file - we already worked out where this is in the Calibration tab, so we shouldn't have to go and find it again
- GSAS-II Installation Directory - it'd be nice to have this in the Settings tab, instead of this one

Also surely there must be a nicer way of laying it out that gives us a bigger plot area

josephframsay, 2018-04-10:
Settings tab
-------

Make sure that all the various settings aren't just clumped together into one massive struct. Apart from that, I think this one should be fairly straightforward (and might be a good one to start with)

josephframsay, 2018-04-10:
General things
-----------

Most of these tabs run `Engg` algorithms asynchronously - if you run them in the GUI thread then you lock Mantid and your users don't think this is cool. In general, I think it's nicer to run algorithms asynchronously in the Model rather than the Presenter, but this introduces the problem that the presenter then doesn't know when to do the next stage after the algorithm has finished. A pattern I think is quite clean for doing this is to have a ModelWorker class, which runs the algorithm in its own thread and then notifies the Model once it is complete (I did this with Qt signals and slots, though some people will tell you that have Qt in the Model is evil). The Model, through a **very** restrictive interface, then notifies the Presenter that the algorithm has either succeeded or failed, and the presenter proceeds accordingly. You can see how this works in the GSAS tab - the Presenter implements the `IEnggDiffGSASFittingObserver` interface, which allows the model to give it notifications about GSASIIRefineFitPeaks. Obviously don't feel you have to follow this pattern - I'm sure there are better ways of doing this

Don't be tempted to just use the ADS as your model and instantiate all your algorithms from the presenter (I know it's tempting) - you just end up with a giant Presenter where you can barely even see the trees, let alone the wood

martyngigg, 2019-09-20:
After some initial investigations we are deciding that a better way forward will be a new GUI. The current one is actually quite unreliable and broken in master so the current tests are not adequate. No one has experience with the code so learning to work with it, attempting to split it apart into a more automatically testable and keep it working feels like it will take as much effort as writing something new. Also the scientists are not that keen on the UI/UX anyway.

The algorithm functionality is still there and working and will be kept. This is just about writing a new set of tabs to coerce input into the given set of algorithms.

closed 2019-09-20 by martyngigg