mantidproject/mantid #15372: Mock up for the new single peak fitting tab of the Engg GUI

FedeMPouzols, 2016-02-18:
Let's discuss this and layout a first and simple UI design for discussions with scientists.
This should essentially be tidy and custom interface to EnggFitPeaks (which might be extended with new inputs or outputs) to support new functionality around the basic single peak fitting.
As first step, have a look at EnggFitPeaks and its algorithm dialog.

FedeMPouzols, 2016-02-18:
This will probably benefit from the MantidWidget::PreviewPlot (#15099) or similar widgets.

shahrozahmed1, 2016-03-02:
@FedeMPouzols Mockup for for single peak fitting tab can be found [here](https://github.com/mantidproject/documents/tree/master/Design/ENGIN-X/GUI).

FedeMPouzols, 2016-03-02:
Good. We have discusses this today with the ENGIN-X scientists (together with #15374) and the plan looked good to them. So let's just go ahead.

For the "pick peak" functionality, the [ALCPeakFitting interface](https://github.com/mantidproject/mantid/blob/master/MantidQt/CustomInterfaces/inc/MantidQtCustomInterfaces/Muon/ALCPeakFittingView.h) has good examples of use of the widgets that probably will cover all our needs: `QwtPlotCurve`, `MantidWidgets::PeakPicker`, and maybe `MantidQt::MantidWidgets::ErrorCurve`.

FedeMPouzols, 2016-03-02:
This issue can be closed now. The mock-up is available from the link given above.

closed 2016-03-02 by FedeMPouzols