mantidproject/mantid #15374: Mock up for a peaks diagnosis dialog

FedeMPouzols, 2016-02-18:
Let's come up with a simple UI design for discussions with scientists.

For the bank (or texture bank / one block of detectors / etc.) calibrations you should be able to see a plot of all the peaks and the TZERO/DIFC/DIFA/DIF? curve (as the calibration tab is starting to generate now), and also:
- a table with parameters / errors / statistics about the TZERO/DIFC... curve
- a plot of the fitting of every individual peak (the individual points in the TZERO/DIFC curve)
- a table with parameters / errors / statistics about the fitting of every individual peak

As sometimes users might be fitting of the order of tens of peaks, some kind of filtering by interesting parameters (like error estimate, fitting Chi^2) etc. is worth considering as well.

For a start this dialog will be used from the Engg GUI, on demand, and also as a validation step when performing calibrations.

The diagnosis for full calibrations is more complicated but should be designed as just a layer on top of the single detector diagnosis. This will probably require more filters, especially because in this case you're dealing with thousands of TZERO/DIFC... curves, each of them with tens of peaks.

shahrozahmed1, 2016-03-02:
@FedeMPouzols Mockup for for calibration diagnosis dialog can be found [here](https://github.com/mantidproject/documents/tree/master/Design/ENGIN-X/GUI).

FedeMPouzols, 2016-03-02:
Good. We have discusses this today with the ENGIN-X scientists (together with #15372) and the plan looked good to them. We got feedback on all the small details. So let's just go ahead.

FedeMPouzols, 2016-03-02:
This issue can be closed now. The mock-up is available from the link given above.

closed 2016-03-02 by FedeMPouzols