mantidproject/mantid #15376: Investigate different compression techniques for creating windows installer with NSIS

ElliotAOram, 2016-02-18:
Building the windows installer with NSIS seems rather slow with the default compressor. Different compressor options are available and should be investigated to attempt to find the optimal size/speed ratio.

One possible route of investigation is the ability to set the dictionary size for compression on the LZMA compressor.

The following test results were acquired during #15206 and are therefore slightly faster than normal until that fix is merged.

| Compression  |  Build Time  | Install Time  |  Size  |   Size on Disk   |  Installer Size  |
|:------------------:|:---------------:|:---------------:|:-------:|:-------------------:|:------------------:|
|  default            |     07:40       |       00:37     |  789   |         821          |        183          |
|  default            |     07:04       |       00:37     |  789   |         821          |        183          |
|        BZIP2       |     03:00       |       00:42     |  789   |         821          |        229          |
|        BZIP2       |     02:59       |       00:43     |  789   |         821          |        229          |
|        LZMA       |     07:00       |       00:37     |  789   |         821          |        183          | 
|        LZMA       |     07:09       |       00:39     |  789   |         821          |        183          |
|         ZLIB        |     03:15       |       00:18     |   789  |         821          |         254         |
|         ZLIB        |     03:21       |       00:19     |   789  |         821          |         254         |

ElliotAOram, 2016-02-18:
Adding the following to line 39 in `buildconfig\CMake\NSIS.template.in` allow you to change the LZMA dictionary size
`SetCompressorDictSize 1`

|   Compression    |  Build Time  | Install Time  |  Size  |   Size on Disk   |  Installer Size  |
|:---------------------:|:---------------:|:---------------:|:-------:|:-------------------:|:------------------:|
| LZMA (1MB dict) |     06:09      |       00:37     |  789   |         821          |        186           |

ElliotAOram, 2016-02-23:
It's worth investigating another compressor option `/SOLID` the following have all been run with the `/SOLID` option enabled. This option means that a certain amount of unpacking is done before the installer runs producing a screen like this 
[image]

|   Compression    |  Build Time  | Install Time  |  Size  |   Size on Disk   |  Installer Size  |unpack time|
|          ZLIB          |      03:28      |      00:13      |  789   |          821          |        247          |    00:05      |
|          ZLIB          |      03:27      |      00:14      |  789   |          821          |        247          |    00:05      |

ElliotAOram, 2016-02-23:
@martyngigg I think based on the data gathered above using `ZLIB` for compression is probably the best in terms of bringing the install time down. The only downside is it does increase the installer size substantially but I think this is slightly offset by the changes in #15206.
Using the `/SOLID` option on the surface appears beneficial but I think it'd be best to potentially test this on other machines to ensure this actually keeps the overall install time the same? (Perhaps this is a separate issue?)

martyngigg, 2016-02-23:
@ElliotAOram The nightly we are currently distributing is 283Mb after your changes in #15206, which is down from ~300Mb for `3.6.0`. Did you have the `PACKAGE_DOCS=ON` cmake option turned on? We should check what size we end up with including the docs and using `ZLIB`. If it's less than 300Mb then  I think we should go ahead.

@NickDraper Any thoughts?

ElliotAOram, 2016-02-24:
Test with `PACKAGE_DOCS=ON` for ZLIB:

|   Compression    |  Build Time  | Install Time  |  Size  |   Size on Disk   |  Installer Size  |
|:---------------------:|:---------------:|:---------------:|:-------:|:-------------------:|:------------------:|
|          ZLIB           |     04:06      |       00:22     |  868   |         907           |       307           |
|          ZLIB           |     03:59      |       00:22     |  868   |         907           |       307           |

martyngigg, 2016-02-24:
Thanks, @ElliotAOram. Those are interesting findings. Could you check what we'd get with the `LZMA` & 1Mb dict with the packages on?

NickDraper, 2016-02-24:
While a smaller installer would be a nice thing, a 20% reduction is not worth a 80% increase in build time.  For me ZLIB looks like the winner.

ElliotAOram, 2016-07-18:
Information in this issue might no longer be up-to-date time wise but will still be relevant in terms of compression ratios between different algorithms.
Users seem to favour shorter download times (at the time this issue was created)

ElliotAOram, 2017-09-20:
Currently we are favouring a smaller package over faster build times. The default configuration with NSIS does this, so therefore this issue can be closed as it is no longer relevant

closed 2017-09-20 by ElliotAOram