mantidproject/mantid #15392: private variable in ISISRunLogs is not used

quantumsteve, 2016-02-21:
I recently saw the following warning from ISISRunLogs.h
```
/Users/svh/Documents/MantidProject/mantid/Framework/DataHandling/inc/MantidDataHandling/ISISRunLogs.h:64:13: warning: private field 'm_numOfPeriods' is not used [-Wunused-private-field]
  const int m_numOfPeriods;
            ^
1 warning generated.
```

This variable is set in the constructor and then never used. Should this variable be removed from the constructor or is something more serious going on?

martyngigg, 2016-02-22:
@quantumsteve It looks like the handling of parsing the total number of periods was moved to `LogParser` in `Kernel` (https://github.com/mantidproject/mantid/blob/4addb372dc0adbfc533d9b57296681406fb5ebd5/Framework/Kernel/src/LogParser.cpp). This actually makes more sense that the number of periods being an input as it should come from a special log that ISIS creates. I think references to that variable can be safely removed and the constructor argument removed, also.

quantumsteve, 2016-02-24:
have clone() return a `unique_ptr` instead of a `shared_ptr`. Unless we need covariant return types, add `make_unique<T>` as a friend, return a `unique_ptr<T>` and see what breaks.

PR #15405 (2016-02-24): Removed unused private field m_numOfPeriods from ISISRunLogs.
This fixes #15392.

No release notes. This addresses a compiler warning found in PR #15327. 

I removed the unused private field `m_numOfPeriods`, and updated the constructor and functions calling it. While working on this class, I also removed all calls to new `new` in favor of `unique_ptr` and `make_unique`

Since there is user-defined constructor, there won't be an implicitly created default constructor. Explicitly deleting it is unnecessary.

code review: Please check I didn't change the behavior of this class.

testing: This fi [... cut]
closed 2016-02-24 by FedeMPouzols