mantidproject/mantid #15394: AlignComponents doc tests fail on rhel6

FedeMPouzols, 2016-02-22:
The 3 examples included in the doc page of new `AlignComponent` have failed: http://builds.mantidproject.org/job/master_doctests/230/label=rhel6-build/testReport/junit/docs/algorithms_AlignComponents-v1/rotation/
There seem to be issues which are probably specific to rhel6:
```
File "algorithms/AlignComponents-v1.rst", line 183, in rotation
Failed example:
    LoadCalFile(InstrumentName="PG3",
          CalFilename="PG3_golden.cal",
          MakeGroupingWorkspace=False,
          MakeOffsetsWorkspace=True,
          MakeMaskWorkspace=True,
          WorkspaceName="PG3")
    ws = LoadEmptyInstrument(Filename="POWGEN_Definition_2015-08-01.xml")
    components="bank26,bank46"
    bank26Rot = ws.getInstrument().getComponentByName("bank26").getRotation().getEulerAngles()
    bank46Rot = ws.getInstrument().getComponentByName("bank46").getRotation().getEulerAngles()
    print "Start bank26 rotation is [{:.3f}.{:.3f},{:.3f}]".format(bank26Rot[0], bank26Rot[1], bank26Rot[2])
    print "Start bank46 rotation is [{:.3f}.{:.3f},{:.3f}]".format(bank46Rot[0], bank46Rot[1], bank46Rot[2])
    AlignComponents(CalibrationTable="PG3_cal",
            Workspace=ws,
            MaskWorkspace="PG3_mask",
            EulerConvention="YZX",
            AlphaRotation=True,
            ComponentList=components)
    ws=mtd['ws']
    bank26Rot = ws.getInstrument().getComponentByName("bank26").getRotation().getEulerAngles()
    bank46Rot = ws.getInstrument().getComponentByName("bank46").getRotation().getEulerAngles()
    print "Final bank26 rotation is [{:.3f}.{:.3f},{:.3f}]".format(bank26Rot[0], bank26Rot[1], bank26Rot[2])
    print "Final bank46 rotation is [{:.3f}.{:.3f},{:.3f}]".format(bank46Rot[0], bank46Rot[1], bank46Rot[2])
Exception raised:
    Traceback (most recent call last):
      File "/usr/lib64/python2.6/doctest.py", line 1253, in __run
        compileflags, 1) in test.globs
      File "<doctest rotation[0]>", line 11, in <module>
        print "Start bank26 rotation is [{:.3f}.{:.3f},{:.3f}]".format(bank26Rot[0], bank26Rot[1], bank26Rot[2])
    ValueError: zero length field name in format
```

```
File "algorithms/AlignComponents-v1.rst", line 127, in position
Failed example:
    LoadCalFile(InstrumentName="PG3",
          CalFilename="PG3_golden.cal",
          MakeGroupingWorkspace=False,
          MakeOffsetsWorkspace=True,
          MakeMaskWorkspace=True,
          WorkspaceName="PG3")
    ws = LoadEmptyInstrument(Filename="POWGEN_Definition_2015-08-01.xml")
    component="bank26"
    print "Start position is",ws.getInstrument().getComponentByName(component).getPos()
    AlignComponents(CalibrationTable="PG3_cal",
            Workspace=ws,
            MaskWorkspace="PG3_mask",
            Xposition=True, ZPosition=True,
            ComponentList=component)
    ws=mtd['ws']
    print "Final position is",ws.getInstrument().getComponentByName(component).getPos()
Exception raised:
    Traceback (most recent call last):
      File "/usr/lib64/python2.6/doctest.py", line 1253, in __run
        compileflags, 1) in test.globs
      File "<doctest position[0]>", line 14, in <module>
        ComponentList=component)
      File "/opt/mantidnightly/bin/mantid/simpleapi.py", line 1093, in fake_function
        " This is an error in the core setup logic of the mantid module, please contact the development team.")
    RuntimeError: Mantid import error. The mock simple API functions have not been replaced! This is an error in the core setup logic of the mantid module, please contact the development team.
```

```
File "algorithms/AlignComponents-v1.rst", line 253, in sample
Failed example:
    LoadCalFile(InstrumentName="PG3",
          CalFilename="PG3_golden.cal",
          MakeGroupingWorkspace=False,
          MakeOffsetsWorkspace=True,
          MakeMaskWorkspace=True,
          WorkspaceName="PG3")
    # Mask banks that don't have calibration data
    MaskBTP(Workspace='PG3_mask', Instrument='POWGEN',
            Bank='22-25,42-45,62-66,82-86,102-105,123,124,143,144,164,184,204')
    ws = LoadEmptyInstrument(Filename="POWGEN_Definition_2015-08-01.xml")
    print "Start sample position is",ws.getInstrument().getSample().getPos().getZ()
    AlignComponents(CalibrationTable="PG3_cal",
          Workspace=ws,
          MaskWorkspace="PG3_mask",
          FitSamplePosition=True)
    print "Final sample position is {:.5f}".format(mtd['ws'].getInstrument().getSample().getPos().getZ())
Exception raised:
    Traceback (most recent call last):
      File "/usr/lib64/python2.6/doctest.py", line 1253, in __run
        compileflags, 1) in test.globs
      File "<doctest sample[0]>", line 15, in <module>
        FitSamplePosition=True)
      File "/opt/mantidnightly/bin/mantid/simpleapi.py", line 1093, in fake_function
        " This is an error in the core setup logic of the mantid module, please contact the development team.")
    RuntimeError: Mantid import error. The mock simple API functions have not been replaced! This is an error in the core setup logic of the mantid module, please contact the development team.
```

FedeMPouzols, 2016-02-22:
@rosswhitfield , @peterfpeterson

FedeMPouzols, 2016-02-22:
These failures definitely didn't happen on rhel7, and are not happening on ubuntu (which also runs a nightly doctest on master with the other wikitests).

rosswhitfield, 2016-02-22:
The AlignComponents algorithm doesn't exist on RHEL6 because `scipy.optimize.minimize` doesn't exist on RHEL6 and the [algorithm isn't subscribed ](https://github.com/mantidproject/mantid/blob/master/Framework/PythonInterface/plugins/algorithms/AlignComponents.py#L491-L493). In the unit test I [check if the algorithm exist](https://github.com/mantidproject/mantid/blob/master/Framework/PythonInterface/test/python/plugins/algorithms/AlignComponentsTest.py#L114-L115) before running and I guess I need to do the same thing for doctests. But I did think that the docs didn't build if it wasn't subscribed?

PR #15424 (2016-02-23): Disable AlignComponents usage example tests because it fails on RHEL6.
This disables AlignComponents doctests because it fails on RHEL6 as seen [here.](http://builds.mantidproject.org/job/master_doctests/231/testReport/)

I created another issue #15425 for future me to reimplement these doctests in a way that works for RHEL6.

**To test:**

No testing required, just look at the changes.

Fixes #15394 

Does not need to be in the release notes.

---

#### Reviewer ####

Please comment on the following ([full description](http://www.mantidproject.org/Individual_Ticket_Testing)):

##### Code Revi [... cut]
closed 2016-02-23 by quantumsteve