mantidproject/mantid #15408: script interpreter window stays onscreen when running unit tests

quantumsteve, 2016-02-22:
I found this on OS X. Not sure about other platforms

- Start MantidPlot and detach the  script interpreter window from the main MantidPlot Window.
- Close MantidPlot
- run `ctest -R MantidPlot`. The script interpreter window will pop up while running these tests.

NickDraper, 2016-06-22:
On windows the outline of the window briefly appears an then goes away, just like several of the graphs appear and then  away.

Is this worth fixing?

quantumsteve, 2016-06-22:
I'm usually working on another task while running the unit tests and having a large white box pop up is distracting. I have been able to work around this issue by keeping the script interpreter window docked, but would like to see this eventually fixed. 

I tried searching for the PR that minimized the ApplicationWindow, but didn't find it. I assume this is a small change for someone more familiar with that portion of the code base, but I'm not quite sure where to start.

NickDraper, 2017-07-03:
duplicate of #13270

closed 2017-07-03 by NickDraper