mantidproject/mantid #15421: Mantid doesn't build with GSL 2

rosswhitfield, 2016-02-23:
Mantid fails to build with GSL 2 which will be on Ubuntu 16.04 and Fedora 24. We will need to support Ubuntu 16.04 for the `3.7` release. Hopefully it is limited to just the curve fitting.

We have a Ubuntu 16.04 [build server](http://builds.mantidproject.org/job/master_clean-ubuntu-16.04/). Feel free to point it at your working branch to use for testing.

The type of errors that are appearing:
```
02:04:46 /home/builder/jenkins-linode/workspace/master_clean-ubuntu-16.04/Framework/CurveFitting/src/Algorithms/Fit1D.cpp: In member function ‘virtual void Mantid::CurveFitting::Algorithms::Fit1D::exec()’:
02:04:46 /home/builder/jenkins-linode/workspace/master_clean-ubuntu-16.04/Framework/CurveFitting/src/Algorithms/Fit1D.cpp:615:29: error: ‘struct gsl_multifit_fdfsolver’ has no member named ‘J’
02:04:46        gsl_multifit_covar(s->J, 0.0, covar);
```

rosswhitfield, 2016-02-25:
Here is an example of a patch required to go from gsl 1 to 2 for qtiplot, from [here](https://projects.archlinux.org/svntogit/packages.git/tree/trunk/qtiplot-gsl2.patch?h=packages/qtiplot&id=c5e8e49355cc1a3bed20604f208a9091ac4cb98b).
```diff
Description: adoptation to gsl2.
Author: Damir Islamov <damir@secretlaboratory.ru>
Last-Update: 2015-12-04

--- a/qtiplot/src/analysis/Fit.cpp
+++ b/qtiplot/src/analysis/Fit.cpp
@@ -126,10 +126,12 @@ gsl_multifit_fdfsolver * Fit::fitGSL(gsl
 			break;
 		}
 	}
-
+	gsl_matrix *J = gsl_matrix_alloc(d_n, d_p);
 	if (status){
-	    gsl_multifit_covar (s->J, 0.0, covar);
+	    gsl_multifit_fdfsolver_jac(s, J);
+	    gsl_multifit_covar (J, 0.0, covar);
 	    iterations = 0;
+	    gsl_matrix_free (J);
 	    return s;
 	}
 
@@ -154,9 +156,9 @@ gsl_multifit_fdfsolver * Fit::fitGSL(gsl
 
 		status = gsl_multifit_test_delta (s->dx, s->x, d_tolerance, d_tolerance);
 	} while (inRange && status == GSL_CONTINUE && (int)iter < d_max_iterations);
-
-	gsl_multifit_covar (s->J, 0.0, covar);
-
+	gsl_multifit_fdfsolver_jac(s, J);
+	gsl_multifit_covar (J, 0.0, covar);
+	gsl_matrix_free (J);
 	iterations = iter;
 	return s;
 }
```

AndreiSavici, 2016-04-06:
```
The following tests FAILED:
	447 - PythonAlgorithms_ComputeCalibrationCoefVanTest (Failed)
	462 - PythonAlgorithms_EnggFitPeaksTest (Failed)
	478 - PythonAlgorithms_IqtFitMultipleTest (Failed)
	488 - PythonAlgorithms_LoadNMoldyn4AsciiTest (Failed)
	494 - PythonAlgorithms_MolDynTest (Failed)
	520 - PythonAlgorithms_TOFTOFConvertTofToDeltaETest (Failed)
	1105 - CurveFittingTest_MultiDomainFunctionTest (Failed)
	1142 - CrystalTest_OptimizeCrystalPlacementTest (Failed)
```

AndreiSavici, 2016-04-06:
The following doc tests have problems:
```
EnggCalibrate-v1.rst
FitPeak-v1.rst
EnggFitPeaks-v1.rst
ConvolutionFitSequential-v1.rst
SaveMDWorkspaceToVTK-v1.rst
```
The last one is failing because I did not build with `MAKE_VATES` option. It is still annoying that the test is run, even when the algorithm is not registered

AndreiSavici, 2016-04-07:
I did not check yet if all the problems in systemtests are related to gsl, but the failed ones are:
```
FAILED:
SANSFileChecking.SANSSwitchIDFTestRunner
SANS2DSlicing.SANS2DMinimalBatchReductionSliced
SANS2DSlicing.SANS2DMinimalSingleReductionSliced
WeightedLeastSquaresTest.WeightedLeastSquaresTest
CalibrateRectangularDetector_Test.PG3CCCalibration
SANSLoadersTest.LoadAddedEventDataSampleTestStressTest
SANS2DMultiPeriod.SANS2DMultiPeriodBatch
SANS2DMultiPeriod.SANS2DMultiPeriodSingle
SANS2DReductionGUI.SANS2DGUIBatchReduction
EnggCalibrationTest.EnginXCalibrateFullThenCalibrateTest
SANS2DLimitEventsTime.SANS2DLimitEventsTime
ISISIndirectInelastic.OSIRISConvFit
ISISIndirectInelastic.OSIRISFuryAndFuryFit
ISISIndirectInelastic.OSIRISIqtAndIqtFitMulti
VesuvioCorrectionsTest.GammaAndMsCorrectWorkspaceIndexOne
VesuvioCorrectionsTest.TestGammaAndMsCorrectWorkspaceIndexTwo
SANS2DBatch.SANS2DBatch
SANS2DBatch.SANS2DNewSettingsCarriedAcrossInBatchMode
SANS2DMultiPeriodAddFiles.SANS2DMultiPeriodAddFiles
SANSQResolutionTest.SANSQResolutionWithGravity
SANSQResolutionTest.SANSQResolutionWithoutGravity
SANS2DReductionGUIAdded.SANS2DReductionGUIAddedFiles
SANS2DReductionGUI.SANS2DGUIReduction
SANS2DReductionGUI.SANS2DMinimalBatchReduction
SANS2DReductionGUI.SANS2DMinimalSingleReduction
SANS2DFrontNoGrav.SANS2DFrontNoGrav
SANS2DFrontNoGrav.SANS2DWithExtraLengthGravity
ISISIndirectInelastic.IRISConvFit
ISISIndirectInelastic.IRISFuryAndFuryFit
ISISIndirectInelastic.IRISIqtAndIqtFitMulti

90% tests passed, 30 tests failed out of 313 (22 skipped)
All tests passed? False
```

rosswhitfield, 2016-04-07:
Looks like you're missing `python-nxs` on your system, see #15655

AndreiSavici, 2016-04-07:
Slightly better:

```
FAILED:
CalibrateRectangularDetector_Test.PG3CCCalibration
WeightedLeastSquaresTest.WeightedLeastSquaresTest
ISISIndirectInelastic.OSIRISConvFit
ISISIndirectInelastic.OSIRISFuryAndFuryFit
ISISIndirectInelastic.OSIRISIqtAndIqtFitMulti
ISISIndirectInelastic.IRISConvFit
ISISIndirectInelastic.IRISFuryAndFuryFit
ISISIndirectInelastic.IRISIqtAndIqtFitMulti
EnggCalibrationTest.EnginXCalibrateFullThenCalibrateTest
VesuvioCorrectionsTest.GammaAndMsCorrectWorkspaceIndexOne
VesuvioCorrectionsTest.TestGammaAndMsCorrectWorkspaceIndexTwo

96% tests passed, 11 tests failed out of 313 (22 skipped)
All tests passed? False
```

mantid-roman, 2016-05-06:
@rosswhitfield @AndreiSavici Could you check if it's better now?

rosswhitfield, 2016-05-06:
👍 `EnggFitPeaksTest` and `IntegratePeakTimeSlicesTest` now pass. 😃 

With `SCDCalibratePanelsTest` did you just comment out the test?

I still have `EnggFitDIFCFromPeaksTest` and `OptimizeCrystalPlacementTest` failing.

mantid-roman, 2016-05-06:
I think I did :disappointed:

mantid-roman, 2016-05-09:
Do you have the error message for `OptimizeCrystalPlacementTest `? It passes for me.

rosswhitfield, 2016-05-09:
That test sporadically fails for me. From [here](http://builds.mantidproject.org/job/master_clean-ubuntu-16.04/47/testReport/junit/CrystalTest/OptimizeCrystalPlacementTest/test_basic/)

```
Stacktrace

Error: Expected (d == 0) up to .3 (0.3000), found (1.0817 != 0)
Error: Expected (d == 0) up to .3 (0.3000), found (1.2182 != 0)
Standard Output

Fit finished. Status=success
Fit finished. Status=success

In OptimizeCrystalPlacementTest::test_basic:
/home/builder/jenkins-linode/workspace/master_clean-ubuntu-16.04/Framework/Crystal/test/OptimizeCrystalPlacementTest.h:137: Error: Expected (d == 0) up to .3 (0.3000), found (1.0817 != 0)
/home/builder/jenkins-linode/workspace/master_clean-ubuntu-16.04/Framework/Crystal/test/OptimizeCrystalPlacementTest.h:137: Error: Expected (d == 0) up to .3 (0.3000), found (1.2182 != 0)
```

NickDraper, 2016-05-20:
This woulod be too risky to include in R3.7 now, moving to R3.8

mantid-roman, 2016-05-24:
Somebody with better knowledge of `OptimizeCrystalPlacement` needs to have a look at `test_basic` and probably redesign it. The fits there are ill-conditioned and I have no idea how to make them stable.

rosswhitfield, 2016-05-24:
If you want to open a pull request for what you have done so far it would be good to get it into master, we can then find someone else to look at `OptimizeCrystalPlacement`.

mantid-roman, 2016-05-24:
This test will have to be disabled

rosswhitfield, 2016-05-24:
No? We aren't using gsl 2 on any of our officially supported platforms yet, so the pull request will pass. We just need to solve this problem before we can support Ubuntu 16.04.

PR #16390 (2016-05-27): Build Mantid with gsl2
Some changes are made to make it work with GSL2. Most of the unit tests are fixed. At least one test still fails (`OptimizeCrystalPlacement`).

**To test:**

Build on Ubuntu 16.04.

Fixes #15421.

It is too early for the release notes.

---

#### Reviewer ####

Please comment on the following ([full description](http://www.mantidproject.org/Individual_Ticket_Testing)):

##### Code Review #####

- [ ] Is the code of an acceptable quality?
- [ ] Does the code conform to the [coding standards](http://www.mantidproject.org/Coding_Standards)? Is it well structured with small  [... cut]
closed 2016-05-27 by FedeMPouzols