mantidproject/mantid #15435: Super-WISH instrument definition

raquelalvarezbanos, 2016-02-24:
From Pascal:

> I think I talked before about getting a super-WISH instrument definition so we could see all the reflections that we are missing. Ideally we would be able to load the actual WISH data onto it but it would havbe much more out of plane coverage. 
One feedback from my users is that it would be nice to give the angles between reflections selected by clicking. This would be the actual angles and also in plane and out of plane angles.

samueljackson92, 2016-07-07:
To further clarify this: this is basically just expanding the instrument definition file for WISH to include a fuller range of angles. Contact him for the specifics.

samueljackson92, 2017-04-11:
With the above two PRs merged this is now finished.

closed 2017-04-11 by samueljackson92