mantidproject/mantid #15437: MDImageWorkspace: implement and expose to Python - umbrella

FedeMPouzols, 2016-02-24:
An umbrella issue for MDImageWorkspace tickets and related ones.

- [ ] #15418 - IMDLeanGeometry
- [ ] #14165 - MDLeanGeometry
- [ ] #15419 - IMDLeanWorkspace
- [ ] ? IMDLeanHistoWorkspace
- [ ] ? MDLeanHistoWorkspace
- [ ] ? IMDImageWorkspace and MDImageWorkspace
- [ ] ? Expose to Python

DimitarTasev, 2017-01-09:
TODO will be moved to an individual project as this will be a major change

DimitarTasev, 2017-03-13:
This is unlikely to be implemented. The default data type for workspaces is double (64bits) which makes it very inefficient to deal with large stack of images.

The problem with having X and E dimensions can be solved by sharing an empty vector across.
The problem with having double as a default type remains to be solved, and might require templating _every_ workspace class.

I am closing this issue, if any works is to be done on this in the future the design has to be rethought and redone to account for the large datasets (~20GB), as it currently does not.

closed 2017-03-13 by DimitarTasev