mantidproject/mantid #15454: FindPeaks is 5-10 times slower now compared with release 3.6 (windows only)

FedeMPouzols, 2016-02-25:
Observed when running the EnginX calibration on windows.
This script:
```
ws = CreateSampleWorkspace(Function="User Defined", UserDefinedFunction="name=LinearBackground, A0=0.3;"
                                          "name=Gaussian, PeakCentre=5, Height=10, Sigma=0.7;"
                                          "name=Gaussian, PeakCentre=30, Height=10, Sigma=0.7;"
                                          "name=Gaussian, PeakCentre=50, Height=10, Sigma=0.7;"
                                          "name=Gaussian, PeakCentre=70, Height=10, Sigma=0.7;"
                                          "name=Gaussian, PeakCentre=90, Height=10, Sigma=0.7;"
                                          "name=Gaussian, PeakCentre=110, Height=10, Sigma=0.7;"
                                          "name=Gaussian, PeakCentre=130, Height=10, Sigma=0.7;"
                                          "name=Gaussian, PeakCentre=150, Height=10, Sigma=0.7;"
                                          "name=Gaussian, PeakCentre=170, Height=10, Sigma=0.7;"
                                          "name=Gaussian, PeakCentre=190, Height=10, Sigma=0.7;"
                                          , NumBanks=1, BankPixelWidth=1, XMin=0, XMax=200, BinWidth=0.1)

table = FindPeaks(InputWorkspace='ws', FWHM='20')

row = table.row(0)

#print row
print "Peak 1 {Centre: %.3f, width: %.3f, height: %.3f }" % ( row["centre"],  row["width"], row["height"])
```

Runs in 0.25s with releae-3.6, whereas it takes more than 3s with a release build of current master.

Reverting the direct changes to FindPeaks and FitPeak since release-3.6 does not help. A bit of `git bisect` or similar will be needed.

Related: #15453.

FedeMPouzols, 2016-02-25:
On debian the example script given in the description of this issue runs in 0.11-0.17s regardless of the Mantid version. 
Traditionally it's been faster than windows with calibrations that use FindPeaks (50% or less run time).

DavidFair, 2018-06-14:
Cannot replicate in Mantid 3.11. Runs in approximately 0.08 seconds on Windows

closed 2018-06-14 by DavidFair