mantidproject/mantid #15457: ARGUS data loading

tom-perkins, 2016-02-25:
Raised in muon meeting 25/2 by Fiona and Peter

1. Loading recent ARGUS data into Mantid using the MuonAnalysis interface doesn't look right. The asymmetry plot looks unphysical - speak to Fiona for more information.

2. "Load current run" doesn't work for ARGUS. At the moment it's a special case that tries to load `\\ndw828\argusdata\current cycle\nexus\argus0000000.nxs` - should be changed to match other instruments, i.e. look in `\\argus\data\autosave.run` for the autosave file name.

tom-perkins, 2016-02-26:
Discussion with Fiona and Adam 26/2:

1. The issue is the selection of which group to plot. Detectors 10-16 on ARGUS are currently faulty so when you:
  - Select "long" as group to plot on Home tab
  - Go to Grouping tab, edit "fwd" row of table to exclude these detectors
  - Go back to Home tab

  then the group being plotted has changed to "fwd" behind your back, because that's the row selected in the table!
"long" should always be the default selection (90% of the time). Keep the user's choice until they explicitly change it.

Example dataset to replicate the issue: `ARGUS00060926.nxs`

2. "Load current run": `ndw828` no longer exists. Change ARGUS to match other instruments as described above. This is needed by the Muon training course - high priority.

3. Typing a run number into the box doesn't work for ARGUS as it is padding with the wrong number of zeros. On my PC it works (three zeros) but on Fiona's it didn't - investigate

4. There is no way to specify faulty detectors in SECI so that this information is saved in the NeXus file. (This isn't a Mantid issue)

tom-perkins, 2016-02-26:
Adam raised another point that is not a Mantid issue:
There is no way to specify faulty detectors in SECI so that this information is saved in the NeXus file along with the grouping. @Anders-Markvardsen do you know who is the best person to forward this information on to?

Anders-Markvardsen, 2016-02-26:
SECI is controls. I would try either Freddie Akeroyd or Matt Clarke.

Optionally you may treat this issue as two issues. One where change Mantid to look in `\\argus\data\autosave.run` and one to deal with the groupping (faulty detector) issue (as comment, are different groupping files (IDFs) for loading of different run files on ARGUS required?)

tom-perkins, 2016-02-26:
I think the scientists would rather exclude the bad detectors for now than create a new IDF. As far as I know it's only recent files that are affected (possibly since the start of this cycle).

tom-perkins, 2016-02-29:
Zero padding for ARGUS: 7 for runs 1-57941, 8 for runs 57942-current
This was updated in `Facilities.xml` for release 3.5 and works on my PC

tom-perkins, 2016-02-29:
Load current data: if this fails, the error message should show the remote path that Mantid is trying to read from.

closed 2016-03-02 by FedeMPouzols