mantidproject/mantid #15458: Indirect Bayes Quasi - Causing Mantid Crash with certain files

ElliotAOram, 2016-02-25:
Vicky has highlighted an issue with BayesQuasi when running the following files:
* iris68291_graphite002_red - sample
* iris68015_graphite002_red - resolution

After initially testing this it seems its a problem with interface and not the algorithm.
This crash can be repeated by loading these two files into the interface and then clicked `Run` with the default options.After a second or so, Mantid will close.

ElliotAOram, 2016-02-25:
It would appear upon further investigation this is actually caused by the the 0s in the workspace. If the cropped data that is input into the Fortran has trailing zeros at the end of the data (highest bins) then the algorithm will fail

SpencerHowells, 2016-02-26:
I tested with this data and used the default energy range -0.5:5.0 and it worked fine. Doing this check is a good idea. What does it do when there are a few zeroes before the last one? This could occur for low statistics data.

ElliotAOram, 2016-02-26:
It seems to run fine as long as the first and last data is non zero. But I'm running a few tests on it to make sure this is the case.

ElliotAOram, 2016-03-01:
This is actually to do with the error being 0, rather than the y data as I originally thought

closed 2016-03-01 by peterfpeterson