mantidproject/mantid #15501: Prototype L2, and Theta Cache

OwenArnold, 2016-03-01:
A Required experiment for the [Geometry Milestone](https://github.com/mantidproject/documents/blob/master/Design/Instrument-2.0/milestone1.md).

Attempt to lazy-calculate common information required for detector based calculations and cache it. 

The output of this will be some performance metrics and the performance test.

OwenArnold, 2016-03-01:
[image]
Results show pretty much what I would expect. We also don't seem to pay much for the initial cache. Results shown are for the new Peformance test suite in `GeometryInfoTestPerformance`

OwenArnold, 2016-03-02:
[image] Updated results with locking

OwenArnold, 2016-03-03:
Caching does not appear to provide a large improvement when the l2 calculations are trivial. However, when several components appear in the path 4-or more, the ability to cache can save a large number of vector operations.

OwenArnold, 2016-03-03:
@SimonHeybrock Any suggestions on other things to try along these lines?

SimonHeybrock, 2016-03-03:
@OwenArnold I think it might make sense to consider this problem as 2 separate aspects:
1. The main contribution to the current high cost is creation of the instrument and detectors, and searching parameter maps, etc., for every single lookup. This should definitely be eliminated.
2. The cost of the actual geometric computation, e.g., calculation of distances between points in space.

OwenArnold, 2016-03-15:
I'm closing this because this prototype has reached a dead-end. Interesting to do, but not worth further investigation. I'm instead focusing on the eliminated parameter map approach prototypes at present.

closed 2016-03-15 by OwenArnold