mantidproject/mantid #15532: Structured Detector XML format

LamarMoore, 2016-03-03:
I would like to propose a format for the StructuredDetector geometry which would allow the creation topologically regular, geometrically irregular detector banks. This is related to issue #14965.

```xml
<component type="fan" idstart="1000" idfillfirst="y" idstepbyrow="300">
	<location y="0">
	</location>
</component>

<!-- Produces a fan element rotated 45 degrees with respect to the first.
So 8 of these for LOKI
-->
<component type="fan" idstart="8500" idfillfirst="y" idstepbyrow="300">
	<location y="0">
		<rot val="45" x="0" y="0" z="1">
		</rot>
	</location>
</component>
.
.
.
<!--
Produces a fan with 24 (6*4) detectors. This will result in detector IDs 1000 - 8200 
as described above.
The winding order will be the same as VTK increase in X then Y
            
           *---------*---------*            
            \        |        /
             *-------*-------*
              \      |      /
               *-----*-----*
                \    |    /
                 *---*---*
                  \  |  /
                    ***
So a shape like the above would produce the following points.
-->
<type name="fan" is="StructuredDetector" xpixels="2" ypixels="4">
    <location x="0.0" y="0.0" z="0.0" />
    <location x="0.0" y="0.0" z="0.0" />
    <location x="0.0" y="0.0" z="0.0" />
    <location x="-0.25" y="1.0" z="0.0" />
    <location x="0.0" y="1.0" z="0.0" />
    <location x="0.25" y="1.0" z="0.0" />
    <location x="-0.5" y="2.0" z="0.0" />
    <location x="0.0" y="2.0" z="0.0" />
    <location x="0.5" y="2.0" z="0.0" />
    <location x="-0.75" y="3.0" z="0.0" />
    <location x="0.0" y="3.0" z="0.0" />
    <location x="0.75" y="3.0" z="0.0" />
    <location x="-1.0" y="4.0" z="0.0" />
    <location x="0.0" y="4.0" z="0.0" />
    <location x="1.0" y="4.0" z="0.0" />
    <location x="0.0" y="0.0" z="0.0" />
    <location x="0.0" y="0.0" z="0.0" />
    <location x="0.0" y="0.0" z="0.0" />
    <location x="-0.25" y="1.0" z="0.0" />
    <location x="0.0" y="1.0" z="0.0" />
    <location x="0.25" y="1.0" z="0.0" />
    <location x="-0.5" y="2.0" z="0.0" />
    <location x="0.0" y="2.0" z="0.0" />
    <location x="0.5" y="2.0" z="0.0" />
    <location x="-0.75" y="3.0" z="0.0" />
    <location x="0.0" y="3.0" z="0.0" />
    <location x="0.75" y="3.0" z="0.0" />
    <location x="-1.0" y="4.0" z="0.0" />
    <location x="0.0" y="4.0" z="0.0" />
    <location x="1.0" y="4.0" z="0.0" />
</type>
````

The VTKStructuredDetector analog produces the following:
[image]

OwenArnold, 2016-03-04:
@LamarMoore  Looks good, but I think that `location` is the wrong element name. As this has been used to refer to the centre point location of the detector and in this case it means something completely different. How about `detectorvertex` or something similar instead.

OwenArnold, 2016-03-31:
Done!

closed 2016-03-31 by OwenArnold