mantidproject/mantid #15534: Review ReflectometryReductionOne workflow diagram

raquelalvarezbanos, 2016-03-03:
While testing #15075 I noticed a couple of issues in the workflow diagram:

- Looking at the diagram it seems that we only apply transmission corrections if the input workspace is not in `lambda` (which I think is not correct)
- I find the transmission correction process a bit unclear
- The last step, after `ConvertUnits` should be `RotateSource` again (we are rotating the source back to its original position).

This issue is to review the workflow diagram and update it if necessary.

MattKing06, 2016-03-21:
@OwenArnold would it be possible to ask you a few questions on this workflow diagram as I believe you have the most experience with the algorithm?

raquelalvarezbanos, 2016-05-05:
@MattKing06: see issue #15565

PR #15723 (2016-04-01): Review RRO Workflow diagram
After a series of changes the workflow diagram for the ReflectometryReductionOne algorithm does not match with the flow of code. This PR aims to change the workflow diagram so that it is representative of the current code for the algorithm.

**To test:**
- Build Qt-help documentation for ReflectometryReductionOne-v1
- Step through the workflow and see that it represents the flow in the code.

Fixes #15534 

---

#### Reviewer ####

Please comment on the following ([full description](http://www.mantidproject.org/Individual_Ticket_Testing)):

##### Code Review #####

 [... cut]
closed 2017-04-10 by PranavBahuguna