mantidproject/mantid #15539: Add system test for LoadCIF

MichaelWedel, 2016-03-04:
There should be a system test that loads some CIF-files and verifies the resulting crystal structures are correct. This requires two tasks:

1. Obtain example CIF-files that can go into the repository.
2. Implement system test.

The examples should include files with anisotropic displacement parameters to check that `U_equiv` is calculated correctly.

MichaelWedel, 2016-05-30:
Now that PyCifRW is an official dependency, this can be implemented. A start could be reading the CIF-files that @FedeMPouzols put into the repository for standard engineering materials.

FedeMPouzols, 2016-05-30:
@MichaelWedel: this sounds like a great plan. The EnginX files may not be the easiest start point though.
Besides using non-standard space group names/symbols, these files seem to have syntax issues that PyCifRW doesn't like. I think that most if not all files have issues. Beware that the files were generated with a tool from GSAS (version 1 I think): GSAS2CIF. So these files may deviate from the standard too much. I didn't look at it enough so as to esimate if it's a matter of fixing the files, ignoring certain "extensions", and/or using options in the loader.

closed 2021-03-01 by stale[bot]