mantidproject/mantid #15558: Imaging functionality and GUI for IMAT (release 3.7)

FedeMPouzols, 2016-03-04:
An umbrella for the IMAT related work that should be ready for release 3.7.
Includes at least:

- [ ] #15437 The MDImage workspace.
- [ ] Deal with rotation options properly in the GUI, and the local/remote scripts.
- [ ] Other improvements for the stack of images viewer/selector: #12477.
- [ ] #14155 more pre-processing outputs in the local/remote reconstruction scripts (and other improvements). GUI and usability for this: #13469.
- [ ] Improvements to FITS algorithms: #14040, #13887
- [ ] Aggregation of energy bands available as Mantid algorithm in addition to the scripts already in place: #13839.
- [ ] Good interface and additional features for energy bands: #13467.
- [ ] Corrections for MCP detector if we get mask information and tests: #13838.
- [ ] Make best use of the MDImage workspace where the code is currently using traditional workspaces.
- Keep up to date with new algorithms and versions of the tomographic reconstruction tools. #12790.
- Deal with the versions of tomographic reconstruction tools on the remote machines/SCARF.
- Modifications or improvements in the algorithms and workflow.
- Instrument definition issues.
- Support scientists with reconstruction benchmarks, algorithms, tests, analysis machine setup, etc.
- Support with files and their logs. In particular address issues with nexus files and files from different cameras/detectors.

NickDraper, 2016-05-20:
Federeico, does this ticket need to be split into R3.7 and later release work?

FedeMPouzols, 2016-05-20:
This one is specific to 3.7, there are two more roadmap tickets for 3.8 and 3.9: #15559, #15560.

FedeMPouzols, 2016-05-20:
And I think that this one can be closed now, so I'm putting it "Reviewe Required".
Still not done: the IMDImage workspace.

closed 2016-05-27 by NickDraper