mantidproject/mantid #15560: Imaging functionality and GUI for IMAT (release 3.9)

FedeMPouzols, 2016-03-04:
An umbrella for the IMAT related work that should be ready for release 3.8.
Medium/high uncertainty about what will be the status and requirements.

- Get all details ready for users.
- Work to support SAVU as one more tomographic reconstruction tool should be possible at this stage.
- Keep up to date with new algorithms and versions of the tomographic reconstruction tools.
- GUI usability improvements as we have more users.
- [Extensions and improvements to the MDImage workspace.
- Support scientists with reconstruction benchmarks, algorithms, tests, analysis machine setup, etc.
- Modifications or improvements in the algorithms and workflow.
- Instrument definition issues.
- Support scientists/users with reconstruction benchmarks, algorithms, tests, analysis machine setup, etc.
- Support scientists/users with files and their logs. In particular address issues with nexus files and files from different cameras/detectors.

FedeMPouzols, 2016-09-22:
Note we probably need to bring into this issue some of the points originally included in the 3.8 release issue: #15559. The list of requirements needs revisiting now that IMAT is starting operations.

DimitarTasev, 2016-10-13:
Merged all issues in #17809. Closing this issue.

closed 2016-10-13 by DimitarTasev