mantidproject/mantid #15564: Event mode filtering and multiperiod data

raquelalvarezbanos, 2016-03-04:
Needs discussion with scientists. 

A fix is needed for event mode filtering and multi-period loading, as well as inclusion of event workspace reduction in table interface.

The initial estimated time for this task was 2w-1m.

I may reassign this to Matt.

raquelalvarezbanos, 2016-03-08:
I had a meeting today with Jos and Max, they are going to get back to me once they decide how the data should be processed (we agreed that they will set up a new meeting once they've made a decision).

We will typically have a histogram workspace for monitors and event mode for data (transmission runs could be either), and they need to discuss how the normalization by the monitors should be performed. Ideally, there will be a pre-processing step where event data are converted to a histogram workspace readable by `ReflectometryReductionOne`.

raquelalvarezbanos, 2017-01-09:
Max and Jos have provided the following scripts they use to analyze event mode data:

From Max: [EventRef.zip](https://github.com/mantidproject/mantid/files/639344/EventRef.zip)
From Jos: [time_slice_routines_OFFSPEC.zip](https://github.com/mantidproject/mantid/files/692969/time_slice_routines_OFFSPEC.zip)

As a first step we are implementing Max's script in https://github.com/mantidproject/mantid/issues/18227.

gemmaguest, 2017-04-10:
I'm closing this because the child issue #18227 is closed now and we don't have any other requirements at the moment.

closed 2017-04-10 by gemmaguest