mantidproject/mantid #15565: Harmonisation accross instruments

raquelalvarezbanos, 2016-03-04:
IDFs, routines and user interface.

This was marked as medium priority, so I'm marking it up against release R3.8. The initial estimated time was 4-6 months.

raquelalvarezbanos, 2016-05-05:
#16024 was fixed as part of this roadmap issue

raquelalvarezbanos, 2016-05-05:
Scientists are gathering information about how they perform the reduction on different instruments. For each instrument, they have their own script they sometimes use as an alternative to our `ReflectometryReductionOne` algorithm. They are investigating the differences between those scripts with the aim of harmonising the workflow. In the meantime I have agreed with Jos that we will carefully look at `ReflectometryReductionOne` to understand and document what it does. In issue #15534 we tried to improve the workflow diagram, but it seems that there are many steps that are not included. One example is the `Rebin` step applied to the `IvsLam` workspace. We have to make sure we fully understand `ReflectometryReductionOne`. Once they have made a decision on a unified reduction process, the different bits in `ReflectometryReductionOne` are likely to be re-arranged.

raquelalvarezbanos, 2016-05-27:
I had a discussion with Max and Jos yesterday. Please, see issue https://github.com/mantidproject/mantid/issues/16455 for a list of the changes they suggested.

raquelalvarezbanos, 2016-07-13:
Scientists at ISIS have agreed on a new version of the Reflectometry GUI. The design document sent by Max Skoda can be found [here](https://github.com/mantidproject/documents/blob/master/Design/Reflectometry/Mantid_SuperGUI_v1.1.pptx).

I have created an umbrella issue to work on the new version of the GUI: https://github.com/mantidproject/mantid/issues/16868

gemmaguest, 2017-04-10:
This issue is now only dependant on #17029 so we are hoping to get this done for 3.10

gemmaguest, 2017-09-14:
The dependency on #17029 has been removed because this does not really affect the move to the new GUI. This has been moved instead to issue #15566 relating to new features for the new GUI.

The only remaining dependency is now #18674 to remove the old GUI when scientists confirm that they are happy with the new GUI.

gemmaguest, 2018-02-26:
The dependency on #18674 has been removed because this has been moved to a separate objective, hence we can now close this issue because the main work for harmonisation is complete.

closed 2018-02-26 by gemmaguest