mantidproject/mantid #15566: New features needed for GUI interface

raquelalvarezbanos, 2016-03-04:
Multi detector processing (2d), live data, polarization analysis.

raquelalvarezbanos, 2016-03-08:
Other improvements mentioned by Jos and Max (Polref GUI):

- The second transmission run does not seem to be used properly
- Scaling of LHS workspace
- Save dialog should work similarly to the old GUI
- It seems that the final rebinning (dQ/Q) is not applied.
- Check how the GUI works. Do we need to re-load everything if, for instance, just the Q min is changed? What loading algorithm are we using? (Ideally we should use LoadISISNexus).
- Wavelength min, max and step default values should be read from the parameter file, instead of using the defaults from `ReflectometryReductionOne` (where WavelengthStep is for instance 0.5).

Further improvements:
- They would like to be able to expand/collapse groups.
- When this is done, they would like to have a "Plot" option as in the old GUI.

Review the above requirements and create new issues if needed.

raquelalvarezbanos, 2016-03-08:
This is now blocked by:

- [ ] https://github.com/mantidproject/mantid/issues/14848

MattKing06, 2016-04-01:
After investigating the issues mentioned above, I have provided a further explanation in the user documentation for ISIS Reflectometry interface so that users can understand why there is no necessary fix for these issues. The changes to documentation can be found in PR #15800

gemmaguest, 2017-04-10:
This issue is now only dependent on #17029 so we are hoping to get it done for 3.10

gemmaguest, 2017-05-25:
This issue has been expanded to include Summation in Q, Q error bars, and polarization analysis. These are required for the new GUI to be widely adopted. The dependencies are therefore now as follows:
- [ ] #17029
- [ ] #18829 
- [ ] #18824
- [ ] #18823

gemmaguest, 2018-03-21:
I'm closing this umbrella issue because the child issues are not related

closed 2018-03-21 by gemmaguest