mantidproject/mantid #15567: Reduction of 2d and off-specular data

raquelalvarezbanos, 2016-03-04:
Needs discussion with scientists.

raquelalvarezbanos, 2016-03-10:
This was closed by mistake. I am reopening it

Anders-Markvardsen, 2017-03-09:
@gemmaguest I have changed the description and title of this issue to include comments I had from Max Skoda today and comments recently from Nina Steinke

gemmaguest, 2017-04-18:
We need to have a detailed discussion with scientists to set out the requirements for this work. From initial discussions the work may include the following:
- 2D detectors
- Non-rectangular pixels
- Detector grouping
- Mask files
- Graphical way to define an area of interest (via a new tab on the GUI)
- Define a background area of interest
- Mask Q regions in time and/or space.
- Conversion to Q

Note that they currently expect to get 2D detectors some time from around Sept 2017 to Mar 2018.

Because of the broad and vague scope at the moment we have put an initial estimate of 1 to 2 years.

gemmaguest, 2020-07-30:
Closing because this is now covered by its child issue #26481

PR #15586 (2016-03-09): Add reference to normalization paper to docs
Description of work. New online journal article about normalization is Mantid is added to documentation.  Also space is put before volume printed with lattice parameters because there is no space between last angle and volume for large unit cells.  Run the UnitCell unit test to check.

**To test look at documentation for MDNormSCD and concept MDNorm.**

Fixes #15567.

Does not need to be in the release notes.

#### Reviewer ####

Please comment on the following ([full description](http://www.mantidproject.org/Individual_Ticket_Testing)):

##### Code [... cut]
closed 2020-07-30 by gemmaguest