mantidproject/mantid #15568: Scripts needed for inspection of scan data

raquelalvarezbanos, 2016-03-04:
Talk to scientists about this one.

raquelalvarezbanos, 2016-03-04:
The initial estimate was 6 months, but is difficult to give a correct estimate as I don't know what this refers to (what scripts, how many, etc)

raquelalvarezbanos, 2016-03-08:
Scan data refers to the process by which scientists align the sample. They typically record measurements at different angles and heigths but this is something they do with OpenGENIE I believe, and they are not even sure they want to do this from Mantid. I am therefore removing the Milestone (both here and in the Mantid Roadmap table, I will also change the priority there to Low).

gemmaguest, 2020-08-06:
Closing as stale because there has been no mention of this again

closed 2020-08-06 by gemmaguest