mantidproject/mantid #15569: Improvements to SliceViewer

raquelalvarezbanos, 2016-03-04:
Improvements required to slice-viewer if it is to be made usable for reflectometry applications (scaling issues, buggy usage, non-editable preview plot).

Medium priority, initial estimate 3-6 months but will depend on the number of improvements needed. Needs discussion with scientists.

raquelalvarezbanos, 2016-03-08:
Some improvements requested by Max and Jos in our meeting today:

- Be able to zoom in and out in the preview plot
- The "width in Plane" is defaulted to 0.1, it was suggested that the default could be for instance 10% of the view.
- They would like to be able to change the axes (log-log, etc) and boundaries
- They would like to be able to customize the output name when exporting.

MattKing06, 2016-04-01:
I've investigated the proposed improvements and I think the only one that doesn't need a fix 
is the "changing axis boundaries" one. This is already implemented in the preview plot and can be performed by changing the values in the top section of the preview plot interface of SliceViewer.

The rest of improvements could be theoretically be implemented, however I will have to consult with @martyngigg and anyone else who knows more about Slice Viewer so we can confirm that these changes will not affect any of the users who currently use the Slice Viewer.

gemmaguest, 2017-10-19:
This was mentioned again by Max. Ideally they would like the plot in the slice viewer to be as fully functional as a normal plot window. @martyngigg are there any plans for work in this area as part of Mantid 4.0?

martyngigg, 2017-10-20:
Our current plan was to port the slice viewer as it stands but there is nothing to stop these improvements happening - the list above doesn't seem to contain anything that would be too challenging.

closed 2021-03-01 by stale[bot]