mantidproject/mantid #15579: parameterize the QSettings subgroup for FitPropertyBrowser

jmborr, 2016-03-04:
`QVariant` settings for class `FitPropertyBrowser` are stored in  subgroup `"Mantid/FitBrowser/"`. When subclassing `FitPropertyBrowser`, `QVariant` settings should be permitted to be stored in a different subgroup.

One possible solution is to add a member `m_qsettings` for `FitPropertyBrowser` that could be initialized with a different string for the subclass.

jmborr, 2016-03-04:
@FedeMPouzols @mantid-roman What do you think of this issue and the proposed solution? I plan to subclass `FitPropertyBrowser` but I want to store the `QSettings` in a different place than `"Mantid/FitBrowser"`

mantid-roman, 2016-03-07:
I think that FitPropertyBrowser should be deprecated and FunctionBrowser should be used instead.

jmborr, 2016-03-07:
@mantid-roman I briefly took a look. The main difference I noticed is the ability to set "global" parameters. Are there other differences with the FitPropertyBrowser?

mantid-roman, 2016-03-07:
FitPropertyBrowser is a complete widget for fitting spectra from matrix workspaces. It's heavy, not very clean and difficult to change. FunctionBrowser on the other hand is a control for building functions only. It can be supplemented by FitOptionsBrowser (or its subclass) to set other fit properties. See MultiDatasetFit as an example of their use.

jmborr, 2016-03-07:
Sounds good. I'll use these in place of FitPropertyBrowser. Thanks for pointing them out.

closed 2016-03-07 by jmborr