mantidproject/mantid #15593: EventList::sortPulseTimeTOF() wrong for weighted case?

SimonHeybrock, 2016-03-08:
`EventList.cpp` function `void EventList::sortPulseTimeTOF() const` (around line 1320):

```cpp
void EventList::sortPulseTimeTOF() const {
  // [...]

  switch (eventType) {
  case TOF:
    std::sort(events.begin(), events.end(), compareEventPulseTimeTOF);
    break;
  case WEIGHTED:
    std::sort(events.begin(), events.end(), compareEventPulseTimeTOF);
    break;
  // [...]
  }
}
```

The `WEIGHTED` case should probably read `std::sort(weightedEvents.begin(), weightedEvents.end(), compareEventPulseTimeTOF);`?

@martyngigg

peterfpeterson, 2016-03-08:
I agree that this appears to be a real bug.

martyngigg, 2016-03-09:
@SimonHeybrock Could I pass this over to you?

PR #15620 (2016-03-09): Fix bug in EventList::sortPulseTimeTOF()
Fixes broken sorting for weighted case.

**To test:**
- Make sure the regression test fails when the bug is reintroduced.
- Code review: Another unrelated test was fixed, and two new tests introduced, make sure those are also good.

Fixes #15593.

No release notes since this did not show up in nature?

---

#### Reviewer ####

Please comment on the following ([full description](http://www.mantidproject.org/Individual_Ticket_Testing)):

##### Code Review #####

- [x] Is the code of an acceptable quality?
- [x] Does the code conform to the coding standard [... cut]
closed 2016-03-09 by AndreiSavici