mantidproject/mantid #15600: Complete VESUVIO transfer from VMS

NickDraper, 2016-03-08:
Elliot please add related tickets to this umbrella ticket

martyngigg, 2016-04-11:
@ElliotAOram I think we need to have separate issues for the core transfer list vs the additional features list as I can see this thing just keeping growing.

Unless something is an urgent bugfix or just makes sense to do earlier then we should tackle all core requirements first before any of the additional requests. Can you spin out a separate issue for the additional features?

ElliotAOram, 2016-04-11:
Good idea @martyngigg #15879

NickDraper, 2016-05-20:
Elliot is this correctly targeted at r3.7?

ElliotAOram, 2016-05-26:
All tasks list above are now complete or have been moved to the next release. This issue can be closed

closed 2016-05-26 by ElliotAOram