mantidproject/mantid #15602: Fix and Speed up project saving

NickDraper, 2016-03-08:
Fix and Speed up project saving

 * Save project as Python script (all graphs, viewers to expose python)
 * Performance options?
 1. Partial Save functionality (select graphs / workspaces then only save what is needed)
 2. Express Save - Only save History from a workspace since it was last loaded, then rerun on Load Project, then regenerate graphs
 3. Complete Save (as currently)

sub tickets
- [ ] #3383

For Student

NickDraper, 2016-05-16:
Specific notes from unscripted testing:

1D plot: quite good, retained zooming errors and text lables.
Waterfall plot: retaind appearance, but lost water fall plot option boxes
2D plot: lost arrows and lines (2D would probably too as well), Lost axis scaling and colour bar log and scaling
qtiplot table: good
3D plot basic recreation ok

samueljackson92, 2016-06-21:
Some more notes on this:
 - Looks like most of the code for this is in the god-like ApplicationWindow. It would be nice to pull it out of there instead of just adding to that class.
 - There appears to be a single exposed method from the ApplicationWindow class (```saveProjectAs()```) which causes mantid to crash if you give it the tilde on OSX, i.e. ```saveProjectAs("~/test.mantid")``` but appears to work otherwise.

samueljackson92, 2016-06-29:
I'm still currently working on #16719. I've managed to get wrangle Mantid so that we can actually test loading and saving automatically. This afternoon I got serious with unscripted testing and manually ran though a whole bunch of scenarios. I've compiled a list of tests below and noted what does/doesn't work. I'm sure there's even more cases which have been missed, but we can add to it as they are found.

All of the below were carried out on Mac OSX 10.11.5 (El Capitan). 

| No. | Description                                                          | Pass/Fail | Notes                                                                                                                                                                                                                                                              | Tickets |
|-----|----------------------------------------------------------------------|-----------|--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------|---------|
| 1   | Save project with workspaces without plotting                        | Fail      | Project is saved, but the open workspace is not included in the project.                                                                                                                                                                                           |         |
| 2   | 1D plot with a single spectrum plotted                               | Pass      | Plot appears on original monitor (yet was positioned on second).                                                                                                                                                                                                   |         |
| 3   | 1D plot with multiple spectra plotted from the same workspace        | Pass      |                                                                                                                                                                                                                                                                    |         |
| 4   | 1D plot with axis labels changed                                     | Pass      |                                                                                                                                                                                                                                                                    |         |
| 5   | 1D plot with title changed                                           | Pass      |                                                                                                                                                                                                                                                                    |         |
| 6   | 1D plotting with two separate plots from two different workspaces    | Pass      |                                                                                                                                                                                                                                                                    |         |
| 7   | 1D plot with zooming applied.                                        | Pass      |                                                                                                                                                                                                                                                                    |         |
| 8   | 1D plot with error bars                                              | Pass      |                                                                                                                                                                                                                                                                    |         |
| 9   | 1D plot with resized window                                          | Fail      | Plot appears back at the expanded size, but the bottom of the plot is cut off. It looks like the window is resized to be the maximum allowed by the screen, but the plot itself retains the saved size. The option “Rescale to show all” is a work around for now. |         |
| 10  | A single 1D plot with two spectra from different workspaces.         | Pass      |                                                                                                                                                                                                                                                                    |         |
| 11  | 1D plot with logarithmic axes                                        | Pass      |                                                                                                                                                                                                                                                                    |         |
| 12  | 1D plot with font formatting changed                                 | Pass      |                                                                                                                                                                                                                                  [... cut ...] is not necessarily essential.                                                                                                  |         |
| 49  | Script Window with multiple scripts open                             | Pass      |                                                                                                                                                                                                                                                                    |         |

NickDraper, 2017-02-09:
Enough of this is done to close the umbrella ticket for this release

closed 2017-02-09 by NickDraper