mantidproject/mantid #15624: Multiply wrong when multiplying two distribution workspaces

SimonHeybrock, 2016-03-09:
Multiply accepts two distribution workspaces without complaining and multiplies them. There does not seem to be any conversion of the data to take this into account, i.e., the output will be garbage, since that data has been scaled with the *square* of the bin-width.

SimonHeybrock, 2016-03-09:
There seem to be some issues with converting from and to distribution, which may be the cause for this, or related. For example, multiplying two histograms and converting the result to distribution still looks like a histogram.

Jon-Taylor, 2016-03-10:
I ran the following test

Load(Filename='/Users/jonathantaylor/Dropbox/Glyde_March_2015/RawData/MAR19851.raw', OutputWorkspace='hist-data', SpectrumMin=2, SpectrumMax=2)
CloneWorkspace(InputWorkspace='hist-data', OutputWorkspace='raw-point-data')
ConvertToDistribution(Workspace='raw-point-data')
Multiply(LHSWorkspace='raw-point-data', RHSWorkspace='hist-data', OutputWorkspace='dist-times-hist')

Load(Filename='/Users/jonathantaylor/Dropbox/Glyde_March_2015/RawData/MAR19851.raw', OutputWorkspace='hist-data', SpectrumMin=2, SpectrumMax=2)
CloneWorkspace(InputWorkspace='hist-data', OutputWorkspace='raw-hist-data1')
Multiply(LHSWorkspace='hist-data', RHSWorkspace='raw-hist-data1', OutputWorkspace='hist-times-hist')

Load(Filename='/Users/jonathantaylor/Dropbox/Glyde_March_2015/RawData/MAR19851.raw', OutputWorkspace='dist-data', SpectrumMin=2, SpectrumMax=2)
CloneWorkspace(InputWorkspace='dist-data', OutputWorkspace='raw-dist-data1')
ConvertToDistribution(Workspace='dist-data')
ConvertToDistribution(Workspace='raw-dist-data1')
Multiply(LHSWorkspace='dist-data', RHSWorkspace='raw-dist-data1', OutputWorkspace='dist-times-dist')

which generates three datasets
**Hist-times-hist -** binwidth normalisation give a starnge result as expected for data with non constant bin widths. option to convert to distribution does not normalise 'correctly', i.e the not the expected way. 
**Dist- times-dist** - as expected no option to normalise by bin width

**dist-times-hist **- unexpected bin width normalisation works as in dist-times-dist and one can cycle the normalisation on and off. overplotting this data with a distribution data confuses the bin width normalisation option on the plot window

SimonHeybrock, 2016-08-05:
See also https://github.com/mantidproject/documents/pull/25.

closed 2021-03-03 by stale[bot]