mantidproject/mantid #15655: SANS SystemTests fail without python-nxs

rosswhitfield, 2016-03-11:
#14918 has added a dependency on `python-nxs` in `SANSUtility.py` which is needed to pass the SystemTests. The following tests fail without it:
```
    SystemTests.SANS2DBatch.SANS2DBatch
    SystemTests.SANS2DBatch.SANS2DNewSettingsCarriedAcrossInBatchMode
    SystemTests.SANS2DMultiPeriod.SANS2DMultiPeriodBatch
    SystemTests.SANS2DMultiPeriod.SANS2DMultiPeriodSingle
    SystemTests.SANS2DMultiPeriodAddFiles.SANS2DMultiPeriodAddFiles
    SystemTests.SANSLoadersTest.LoadAddedEventDataSampleTestStressTest
    SystemTests.SANS2DSlicing.SANS2DMinimalBatchReductionSliced
    SystemTests.SANS2DSlicing.SANS2DMinimalSingleReductionSliced
    SystemTests.SANSQResolutionTest.SANSQResolutionWithGravity
    SystemTests.SANSQResolutionTest.SANSQResolutionWithoutGravity
    SystemTests.SANS2DReductionGUI.SANS2DGUIBatchReduction
    SystemTests.SANS2DReductionGUI.SANS2DGUIReduction
    SystemTests.SANS2DReductionGUI.SANS2DMinimalBatchReduction
    SystemTests.SANS2DReductionGUI.SANS2DMinimalSingleReduction
    SystemTests.SANS2DReductionGUIAdded.SANS2DReductionGUIAddedFiles
    SystemTests.SANS2DLimitEventsTime.SANS2DLimitEventsTime
    SystemTests.SANS2DFrontNoGrav.SANS2DFrontNoGrav
    SystemTests.SANS2DFrontNoGrav.SANS2DWithExtraLengthGravity
    SystemTests.SANSFileChecking.SANSSwitchIDFTestRunner
```

`python-nxs` is not a dependency of mantid and so system tests shouldn't depend on it. This system tests need to be rewritten to not use `python-nxs` or to skip the tests when `import nxs` fails :disappointed: 

If we are going to add a new dependency to mantid I would prefer to add `python-h5py` instead of `python-nxs` but this would require a larger discussion first.

picatess, 2016-03-14:
@rosswhitfield : It appears that `python-nxs` is a dependency of Mantid. See [here](https://github.com/mantidproject/mantid/blob/master/CMakeLists.txt#L254) and [here](https://github.com/mantidproject/mantid/blob/master/CMakeLists.txt#L296). Is my assumption not correct?

rosswhitfield, 2016-03-14:
So it's an install time dependency. My bad. But `python-nxs` is NOT bundled as part of the OSX packages and this needs to be fixed.

I was running the system tests on my local build and `python-nxs` is not included in our developer packages ([here](https://github.com/mantidproject/mantid/blob/master/buildconfig/dev-packages/deb/mantid-developer/mantid-developer-1.2.6/DEBIAN/control) and [here](https://github.com/mantidproject/mantid/blob/master/buildconfig/dev-packages/rpm/mantid-developer/mantid-developer.spec)) so these all failed for me.

I'm still unsure why  `python-nxs` was ever added as a dependency as it appears this pull request was the first to ever use it (maybe it's previous uses have since been removed). Unless I'm missing something. It does seems strange to have a dependency that is only required for the system tests and nothing else.

picatess, 2016-03-14:
The dependency is being used by the ISIS SANS code in two ways: 
  * To check the `start_time` in a Nexus file
  * To check if a file contains an event workspace
In both cases we want this metadata before loading the data.

rosswhitfield, 2016-03-14:
Previously, before this pull request, you weren't using it or you just didn't test its usage? Because this is the first time it's been an issue.

rosswhitfield, 2016-03-14:
@quantumsteve looks like you need to add `python-nxs` to the OSX bundle now.

rosswhitfield, 2016-03-14:
Previously it looks like `nxs` was only used in [can_load_as_event_workspace](https://github.com/mantidproject/mantid/blob/1fa299e131090ac0b8960d1d7747729560b2dcb2/scripts/SANS/SANSUtility.py#L1315) but you had a [check](https://github.com/mantidproject/mantid/blob/1fa299e131090ac0b8960d1d7747729560b2dcb2/scripts/SANS/SANSUtility.py#L27) first that you can import `nxs`. Why can't this check be added to the new stuff?

picatess, 2016-03-14:
1. The check for event workspaces did have [tests](https://github.com/mantidproject/mantid/blob/master/Testing/SystemTests/tests/analysis/SANSWorkspaceTypeTest.py). 

2. The newly added check for the `start_time` does have the same [check](https://github.com/mantidproject/mantid/blob/master/scripts/SANS/SANSUtility.py#L1578:L1596). 

What message do you get from your system test failure?

rosswhitfield, 2016-03-14:
Getting a lot of things like `AssertionError: '' != '2012-10-22T11:14:54'`

picatess, 2016-03-14:
Hmm, yes that would be expected if OSX does not provide the dependency. Making the tests silent does not solve the actual issue of potential users not having this dependency on OSX. As Steve suggested, it might be best to move to `h5-py` to resolve this issue

AntonPiccardoSelg, 2017-10-05:
This has been sorted out by Pete. He replaced it with `h5py`

closed 2017-10-05 by AntonPiccardoSelg