mantidproject/mantid #15672: Generalise ReflTableViewPresenter

raquelalvarezbanos, 2016-03-15:
Blocked by:

- [ ] #14921

Implement steps 4, 5 and 6 in the [design document](https://github.com/mantidproject/documents/blob/master/Design/DataProcessorAlgorithmUI/DataProcessingUserInterface.md#implemenation-steps):

4. Refactor `ReflTableViewPresenter` so that it creates and configures the DataProcessingAlgorithm completely via it's arguments. Then rename it to `GeneralDataProcessorPresenter`. The `GeneralDataProcessorPresenter` should contain nothing that is relfectometry specific. This will also mean abstracting the post processing and pre processing steps.
5. Make `GeneralDataprocessorPresenter` take all arguments as abstract behaviours.
6. Move `GeneralDataProcessorPresenter`, the views and the behaviours into the MantidQtAPI package.

raquelalvarezbanos, 2016-04-05:
Our concrete table presenter, `ReflTableViewPresenter`, (which will be renamed to `GenericDataProcessorPresenter`, according to the design document), has a method, `reduceRow(int rowNo)`, that:

1. Loads and sums the runs
2. Creates the transmission workspace
3. Runs `ReflectometryReductionOneAuto`
4. Runs `Scale` on `IvsQ`
5. Runs `Rebin` on `IvsQ`

Items 1 and 2 are part of the pre-process step. 

**1. Load and sum the runs**

I think this pre-processing step is quite general. All we need to do is specify how we want to combine the runs. In the case of Reflectometry, something like:

```
pre_process = { 'algorithm':'Plus', 'column_name':'Run(s)', 'direct_output_to':'InputWorkspace'}
```

should be enough. The presenter would load the runs (if necessary, they may already exist in the ADS and in this case we would not need to load them again), and then apply the specified algorithm, `Plus` in this example. To avoid loading runs existing in the ADS, we will also have to specify a prefix, so the above `pre_process` could be something like:

```
pre_process = { 'algorithm':'Plus', 'column_name':'Run(s)', 'direct_output_to':'InputWorkspace', 'prefix':'TOF_'}
```

We may also need to indicate the name of the input properties:

```
pre_process = { 'algorithm':'Plus', 'column_name':'Run(s)', 'direct_output_to':'InputWorkspace', 'prefix':'TOF_', 'direct_input_to':'LHSWorkspace, RHSWorkspace'}
```

**2. Make the transmission workspaces**

Something similar should work.

```
pre_process = { 'algorithm':'CreateTransmissionWorkspaceAuto', 'column_name':'Transmission Run(s)', 'direct_output_to':'FirstTransmissionRun', 'prefix':'TRANS_', 'direct_input_to':'FirstTransmissionRun, SecondTransmissionRun'}
```
Again, the presenter would load the runs (if necessary) and combine them using the specified algorithm, `CreateTransmissionWorkspaceAuto`.

**ReflectometryReductionOneAuto + Scale + Rebin**

These form the processing step. I think the easiest thing to do would be to add the `Scale` and `Rebin` steps to `ReflectometryReductionOneAuto`. If we did this, all we would have to do would be something like this:

```
data_processor_algorithm = 'ReflectometryReductionOneAuto'
```

Then the presenter would use the existing columns as input properties to this algorithm (it would also apply any properties specified in the `Options` column). There are two reasons why I think this would be the best approach: first, we would only need to specify one processing algorithm. If they are kept as separate algorithm, our `data_processor_algorithm` would have to be something like:

```
data_processor_algorithm = ['ReflectometryReductionOneAuto', 'Scale', 'Rebin']
```

and in addition, we would have to find a way to specify which of the output properties from `ReflectometryReductionOneAuto` have to be used as input to `Scale` and which of the output properties from `Scale` should be used as input to `Rebin`.
The second reason, is that scientists have mentioned a couple of times that they would like to be able to specify the `Q` range (`Qmin`, `Qmax` and `Qstep`) in `ReflectometryReductionOne`, which is related to the last `Rebin` step. If they want to have this `Rebin` step as part of the reduction process, I can't see why we couldn't also add the `Scale` step. Maybe there is a good reason not do this.

OwenArnold, 2016-04-07:
For 1. I like the syntax. However, the use of the ADS is a work-around. Ideally we should not have to use it at all. Therefore not excessively tie ourselves to in particular managing existing entries if possible.

Also, I don't understand this: ```'direct_input_to':'LHSWorkspace, RHSWorkspace'```. Could you explain that?

For 2. Ok. We could add these to Reflectometry Reduction one, just remember that Scale is optional (maybe just default it to 1) and so is Rebin, but we should not apply the rebin at all unless instructed to do so.

raquelalvarezbanos, 2016-04-12:
@OwenArnold, I have a question about the post-processing step: 

How should we specify the post-processing algorithm's input properties? 

In Reflectometry, our post-processing algorithm is `Stitch1DMany`, and the input properties `StartOverlaps`, `EndOverlaps` and `Params` are mandatory. I see two possible answers to the question:

**1. Using table columns**

This is how the interface currently works: to stitch a group, we read `Q min`, `Q max` and `dQ/Q` from the table, and use those values for the stitching. In Reflectometry this is fine, because `Q min`, `Q max` and `dQ/Q` are used for the reduction too. But maybe in a different area we would need a table with too many columns (not only columns used for the reduction but also columns used for the post-processing). Besides, we would we mixing different information in the table: information (columns) related to the reduction and information (columns) related to the post-processing.

**2. Using the "Options" menu**

The "Options" menu opens an "Options" dialog. A new tab could be added (this would be similar to what Matt has done in [this](https://github.com/mantidproject/mantid/pull/15677) PR) to specify the options for the post-processing.

The advantage of 2 is that we only have to specify the input properties once for every row, i.e. all rows will share the same value for the input properties. For instance, in Reflectometry, if the input property `ScaleRHSWorkspace` is set to `true`, all groups will be stitched with that property set to true. This may be a disadvantage at the same time, if we wanted/needed to use different values depending on the group, for instance we needed to specify different `StartOverlaps` and/or `EndOverlaps`  for different groups.

Which of the two options would be better? I like the second one more but I guess the answer will depend on how frequently we need different stitching properties for different groups. What do you think?

raquelalvarezbanos, 2016-04-14:
Two variants of what we discussed on Tuesday:

[image]

[image]

The input properties to pre-processing, processing and post-processing algorithms can now be specified via a `HintingLineEdit`. I personally prefer the first option, but what do you think? I can try more alternatives if you don't like any of them.

OwenArnold, 2016-04-15:
@raquelalvarezbanos. Very cool! I also prefer 1.

OwenArnold, 2016-04-15:
@raquelalvarezbanos Is there a PR for this yet? Assign it to me when you create it.

raquelalvarezbanos, 2016-04-18:
Not yet. I still have to test Matt's [PR](https://github.com/mantidproject/mantid/pull/15909), where `Scale` and `Rebin` will be moved to RRO, make sure everything works as before and add the necessary unit tests. I will probably create a PR before writing the tests so we can discuss the design and see if there is anything that could have been done better (which I am sure there will be)

OwenArnold, 2016-04-18:
@raquelalvarezbanos How long do you estimate this phase of the work (this issue) to take?

raquelalvarezbanos, 2016-04-18:
@OwenArnold I have a question about `CreateTransmissionWorkspaceAuto` and `ReflectometryReductionOneAuto`:

While working on this issue today I realized that the interface fails to reduce a run with transmission runs if no processing instructions are specified. As an example, the following reduction:

[image]

fails because of `StrictSpectrumChecking=True` (the error message is that spectrum maps do not match up, and it does not matter if we have two transmission runs or just one). To look into this, I have tried the following script:

```
offspec = Load(Filename='OFFSPEC00010791.raw', PeriodList='1')
trans = CreateTransmissionWorkspaceAuto(AnalysisMode='MultiDetectorAnalysis', FirstTransmissionRun=offspec)
IvsQ, IvsLam, theta = ReflectometryReductionOneAuto(InputWorkspace=offspec, AnalysisMode='MultiDetectorAnalysis')
```

To compare how `ReflectometryReductionOneAuto` and `CreateTransmissionWorkspaceAuto` work. If I inspect the history of `IvsQ` I can see that, when no processing instructions are specified, `ReflectometryReductionOneAuto` analyses all the workspace indices (`ProcessingInstructions=4:244`, as shown below):

[image]

so that the output workspace has (in this particular example) 241 histograms (spectrum numbers from 5 to 245). However, looking at the history of `trans`, I see that `CreateTransmissionWorkspaceAuto` only analyses the first and last spectra when no processing instructions are specified (`ProcessingInstructions=4,244` in the history below)

[image]

So if I modify my script above to use `trans` as the `FirstTransmissionRun` in `ReflectometryReductionOneAuto`, like this:

```
offspec = Load(Filename='OFFSPEC00010791.raw', PeriodList='1')
trans = CreateTransmissionWorkspaceAuto(AnalysisMode='MultiDetectorAnalysis', FirstTransmissionRun=offspec)
IvsQ, IvsLam, theta = ReflectometryReductionOneAuto(InputWorkspace=offspec, AnalysisMode='MultiDetectorAnalysis', FirstTransmissionRun=trans)
```

the reduction fails because of the spectrum maps (unless we set `StrictSpectrumChecking=False`, but I assume in most cases we don't want to do that). My question is, why does `CreateTransmissionWorkspaceAuto` consider only the first and last spectra when no processing instructions are given? Is this intended or is it a bug? I would have expected both algorithms, `ReflectometryReductionOneAuto` and `CreateTransmissionWorkspaceAuto`, to behave similarly in terms of processing instructions.

OwenArnold, 2016-04-19:
Sounds like a bug that ought to be fixed.

raquelalvarezbanos, 2016-04-19:
Ok, thanks, I'll create an issue to fix it.

raquelalvarezbanos, 2016-04-19:
@OwenArnold I was looking into how to generalize `ReflGenerateNotebook`. Correct me if I am wrong but it seems to me that all we are doing is reproduce the history of the result workspaces (we also print a summary of the reduction table, but this is very easy to generalize, and we also show instructions for plotting the result workspaces, which should not be too difficult either). If this is correct, I was thinking that maybe an easier approach would be to retrieve the processed workspaces from the ADS and from them, get the history and print it to the notebook. I think this should work, but it is a completely different approach from what we currently have.

PR #15943 (2016-05-16): Generalise ReflTableViewPresenter
First version of the general Data Processor UI.

**To test:**

Fixes #15672.

---

#### Reviewer ####

Please comment on the following ([full description](http://www.mantidproject.org/Individual_Ticket_Testing)):

##### Code Review #####

- [ ] Is the code of an acceptable quality?
- [ ] Does the code conform to the coding standards? Is it well structured with small focussed classes/methods/functions?
- [ ] Are there unit/system tests in place? Are the unit tests small and test the a class in isolation?
- [ ] If there are changes in the release notes then do t [... cut]
closed 2016-05-16 by FedeMPouzols