mantidproject/mantid #15693: Event Sample Temperature Graph has Too Low Precision for Range in Temperature

KarlPalmen, 2016-03-17:
Load CNCS_7860_event.nxs, sample logs, examine SampleTemp.

A graph is plotted with Y-axis numbers, 
279.9, 279.9, 279.9. 280, 280, 280.
The statistics show a range of 279.904 to 279.987.

It is evident that the precision of the temperatures shown on the Y axis should be increased for such a small range. This may be a general problem with plotting.

KarlPalmen, 2016-04-15:
I've observed the same problem in another log: phase2 of REF_L_32035.nxs, which shows time in microseconds as a Y value with a narrow range of 15444.6 to 15448.8 and all Y values in the plot rendered as 1.544e+04.

NickDraper, 2017-06-29:
[image]

The issue is the lack of precision on the Y axis labels

giovannidisiena, 2019-08-16:
As this is a non-trivial issue with QtiPlot code (MantidPlot) which we are not going to continue to support I am going to close this issue (under @Mantid-Matthew direction).

closed 2019-08-16 by giovannidisiena