mantidproject/mantid #15701: Crystal field: multispectral fitting function.

mantid-roman, 2016-03-18:
Write a multi-dataset fitting function to fit multiple spectra. The crystal field parameters will be global others (such as temperature, etc, line shape parameters) will be local.

This is a part of #15684

mducle, 2016-05-12:
@mantid-roman The example on a fake dataset works for me, but the real datasets I'm using (with 2 temperatures and two Ei's) is proving a bit more difficult... I'll see if I can get it to fit tomorrow.

mantid-roman, 2016-05-13:
@mducle There are no scaling factors yet. How do we account for different Ei's?

mducle, 2016-05-13:
@mantid-roman I'm just going to scale the dataset before hand, but eventually I think we'd need a different scaling factor for each Ei. Unfortunately, I'm not even there yet... I keep crashing Mantid, and getting `GSLVector index = 22 is out of range = 22 in GSLVector.get()` errors - because I want to put constraints on all the FWHM of the peaks, and I guess the number of peaks changes if the CEF parameters change too much. Can we add a constraint which can be applied to the FWHM of all peaks? (something like `ToleranceIntensity` or `ToleranceEnergy` - maybe `MinFWHM` and `MaxFWHM`?)

mantid-roman, 2016-05-13:
@mducle What kind of constraint: fixed number or a formula?

mducle, 2016-05-13:
@mantid-roman Just fixed number at the moment - I was finding that for some peaks I was making the FWHM (or sigma) very broad (>1000meV) or very narrow and wanted to restrict this. The dataset I'm testing at the moment is not very good - it's at high temperature so there are lots of overlapping peaks, and it's got a weird background...

NickDraper, 2016-05-20:
@mantid-roman What is the status of this issue?  Is it correctly targeted at R3.7?

mantid-roman, 2016-05-20:
@NickDraper I should move it to 3.8

closed 2016-06-24 by AndreiSavici