mantidproject/mantid #15703: Crystal field: add form factor correction

mantid-roman, 2016-03-18:
Some of the crystal field fitting functions should be able to perform the form factor correction.

This is a part of #15684

mducle, 2016-09-11:
@mantid-roman As discussed in the meeting, this issue will be better handled by an algorithm separate from the crystal field fitting. I've hacked together an algorithm which uses the tables coded into `MagneticIon`. I'll get together some tests for it, and then open a PR.

(There is also an associated issue, #17457 to extend `MagneticIon` to include 5d ions and the dipole formula for rare earths).

mantid-roman, 2016-09-13:
@mducle Could you also add a docs rst page at `docs/source/algorithms`?

mducle, 2016-09-13:
Yes. Will do... just trying to get the example to work... The dataset I want to use is for PrAl3 but the form factor of Pr3+ is not included in the tables (it's not in the international tables for crystallography for some reason - but it's in McPhase and also the tables on the ILL webpage https://www.ill.eu/sites/ccsl/ffacts/ffactnode7.html and I'm trying to hunt down where those numbers actually come from).

closed 2016-09-19 by martyngigg