mantidproject/mantid #15730: Vesuvio - Vesuvio Resonance Algorithm

ElliotAOram, 2016-03-23:
The instrument scientist on Vesuvio have requested an option to load data and sum all the periods together with a normalisation by the monitor.

One implementation of this would be to add this a separate algorithm. Another option is to have it as another mode of operation in LoadVesuvio.

martyngigg, 2016-03-23:
@ElliotAOram Doesn't the current algorithm already normalize by the monitor?

ElliotAOram, 2016-03-23:
@martyngigg It does but I think the emphasis is more on just summing all the periods together (e.g. no differencing at the end). I think it'd be good if I understand the use case for this more before implementing. I'll talk to Matthew and Giovanni to find out more details.

martyngigg, 2016-03-23:
@ElliotAOram I see. If that is the case then it seems like the best way is probably extending `LoadVesuvio`.  We could do that by simply having an extra Difference mode in the list of `None`?

ElliotAOram, 2016-03-23:
I agree that sounds like the most sensible option.

martyngigg, 2016-03-23:
I think we also need to think about refactoring the `LoadVesuvio` code into some separate classes/modules too. There is already quite a lot of different logic in there. With this option we'll basically have 3 possbible mode types:
* `FoilIn`/`FoilOut` mode for diffraction
* `Single`/`Double`/`Thick` modes for differencing
* `None` for plain summing

and I think it would be good to try and keep the code separate for doing these things.

ElliotAOram, 2016-03-23:
I think that'll make things a lot clearer going forward. I'll write up an issue for this

closed 2021-03-03 by stale[bot]