mantidproject/mantid #15731: Vesuvio - Format of commands script output workspaces

ElliotAOram, 2016-03-23:
### Expected behaviour
* To be discussed with the instrument scientist further but to have fewer workspaces put into the ADS while running the user script (driver for `scripts\inelastic\vesuvio\commands.py`)

### Actual behaviour
* Outputs an unnecessarily are large number of workspace.

### Steps to reproduce the behaviour
* Run the Vesuvio driver script

### Platforms affected
* All

DanNixon, 2016-04-02:
Setting `flags['output_verbose_corrections'] = True` in the driver script significantly reduces the number of workspaces created (you get no individual workspaces for each correction).

ElliotAOram, 2016-04-04:
@DanNixon thanks for letting me know. I'll take a look and discuss it with Matthew and Giovanni to see if they are happy with the amount/way the data is output.

ElliotAOram, 2016-08-01:
Matthew and Giovanni seem happy with the new reduced size and have been made aware of the `output_verbose_corrections` flag

closed 2016-08-01 by ElliotAOram