mantidproject/mantid #15738: MuonAnalysis: refer to periods by name

tom-perkins, 2016-03-23:
Created from point 7 in #13096, mentioned again by scientists in meeting 23/3/16.
They would like to be able to refer to periods by name rather than just by number:

> Periods can have names as set up in the DAE control in SECI, e.g. “Light on” or “RF off”. Can these appear in the Interface selector buttons in addition to numbers 1 to N?

This also applies to the prospective simultaneous fitting control on the Data Analysis tab (when selecting which periods to plot).

tom-perkins, 2016-07-18:
In a muon NeXus v1 file, the period names are found in `run/instrument/beam/period_labels`. 

They are stored separated by `;`, for example in `MUSR00015189.nxs` (2 periods) this entry reads `RF on;RF off`.

In a muon NeXus v2 file, the period names are under `run/run/period_labels` (optional) in a similar format. The separator is not necessarily `;` and is given by the attribute `separator`.

tom-perkins, 2016-10-14:
This has been brought up several times by muon scientists. Marking as 3.9

tom-perkins, 2016-11-01:
Finding the period names in the file is OK, see above. What needs consideration here is how they will be shown in the interface - at the moment you can enter a combination like "1+2-3". Discuss with scientists how this should work in the UI.

tom-perkins, 2016-11-04:
One idea is to put a key of names on the front tab e.g.
1: RF on
2: RF off

AnthonyLim23, 2021-02-15:
@Matt-Cumber this might be helpful later

Matt-Cumber, 2021-04-01:
With the addition of #30939 this shouldn't be a problem anymore

closed 2021-04-01 by Matt-Cumber