mantidproject/mantid #15749: GroupDetectors "Grouping Pattern" property does not work properly

MattKing06, 2016-03-24:
### Expected behavior
- Group spectra with the workspace indices corresponding to Grouping Pattern i.e
`GroupingPattern = "1,2"` should return a workspace containing the spectra at the second and third position in the workspace. 
- If we have a workspace that contains `sp-1, sp-3` only and `GroupingPattern = "0,1"` then we should get the same workspace back as an Output.
### Actual behavior
- If we have a workspace that contains `sp-1, sp-3` only and `GroupingPattern = "0,1"` then the algorithm fails to run and a warning is given stating "Spectra number 2 does not exist in workspace"

### Steps to reproduce the behavior
- Load any dataset that has 3 or more spectra using LoadNexus
- In the LoadNexus dialog specify a spectrum list that contains a gap i.e "1,3" or "3,7" etc.
- Bring up the algorithm dialog for `GroupDetectors` with the workspace from LoadNexus as the Input Workspace
- In the "Grouping Pattern" Property enter the same value you did for LoadNexus ("1,3" for example)
- See that the algorithm produces an error complaining about not be able to find the spectra between the two numbers you provided.
### Platforms affected
- All

gemmaguest, 2017-03-30:
I've hit this problem while testing #19252 . I think the problem is [here](https://github.com/mantidproject/mantid/blob/master/Framework/DataHandling/src/GroupDetectors2.cpp#L1297). There is a +1 on this line that is converting the 0-based index to a 1-based number, which is later assumed to be the spectrum number.

It might be possible to refactor this code to return a vector, rather than a buffer that then has to be re-parsed. It would also be worth checking if there are other places where this +1 conversion occurs.

Here is a simple script to reproduce the problem on a cropped workspace:
```
def printVal(ws, i):
    print "Index=" + str(i) + ", Spectrum=" + str(ws.getSpectrum(i).getSpectrumNo()) + ", Value = " + str(ws.readY(i))

sample = CreateWorkspace(DataX=[1], DataY=[1,2,3,4,5,6,7,8,9,10], NSpec=10)
print "Original workspace:"
for i in range(0,10):
    printVal(sample,i)
print

group1 = GroupDetectors(sample, GroupingPattern="4-5")
print "Grouped original workspace with GroupingPattern='4-5'. This sums rows 4-5 => spectra 5-6:"
printVal(group1,0)
print

cropped = CropWorkspace(sample, StartWorkspaceIndex=4)
print "Cropped workspace:"
for i in range(0,6):
    printVal(cropped,i)
print

group2 = GroupDetectors(cropped, GroupingPattern="4-5")
print "Grouped cropped workspace with GroupingPattern='4-5'. This sums rows 0-1 => spectra 5-6 (but expected rows 4-5 => spectra 9-10):"
printVal(group2,0)
print

group3 = GroupDetectors(cropped, WorkspaceIndexList="4-5")
print "Grouped cropped workspace with WorkspaceIndexList='4-5'. This sums rows 4-5 => spectra 9-10:"
printVal(group3,0)
print
```

gemmaguest, 2017-04-07:
I'm going to look at this as part of #18829

closed 2017-05-05 by martyngigg