mantidproject/mantid #15750: GroupDetectors allows SpectraList to contain too many spectra

MattKing06, 2016-03-24:
### Expected behavior
- When GroupDetectors is provided with more spectra than the amount in a workspace the algorithm should fail and report back to the user that those spectra do not exist.
### Actual behavior
- Algorithm finishes and a workspace containing one spectra with the sum of all the available spectra as an entry in Y values.
- This spectrum is now labelled as `sp-0`
### Steps to reproduce the behavior
- Load in any run file
- Open dialog for GroupDetectors and use this workspace as the InputWorkspace
- Provide a list of spectra for the SpectraList property that exceeds the number of spectra in the workspace
- Run GroupDetectors
- See that the algorithm finishes and the one spectrum in the output workspace contains the sum of all of the spectra in the SpectraList.
- See that this spectrum is labelled as `sp-0` in the data view.
### Platforms affected
- All

SimonHeybrock, 2016-03-29:
Have a look at https://github.com/mantidproject/documents/pull/13#issuecomment-199483356. If we go forward with that suggestion it may implicitly fix this issue, so it might be worth postponing work on this issue for now.

PranavBahuguna, 2016-08-22:
I can no longer replicate the behaviour described by this issue. Considering closing this.

closed 2016-08-22 by PranavBahuguna