mantidproject/mantid #15777: InstrumentView Widget Color Map not updating with streamed live data

lottiegreenwood, 2016-03-30:
### Expected behavior
When run with the attached test script, the colour map on the instrument view should change as the streamed data updates. 
### Actual behavior
However the colours remain the same, despite the spectra plot updating. The colour map will update if you manually click within the maximum/minimum x value number input box and then click on the button for resizing to change the integration range.
### Steps to reproduce the behavior
Run the below script with a developer branch version of mantidpython. 
The POWDIFF_Definition.xml is located here: https://github.com/DMSC-Instrument-Data/live-data-prototype/tree/master/data

lottiegreenwood, 2016-03-30:
``` python
from threading import Thread
import time
import sys
import PyQt4
import mantid.simpleapi as simpleapi
import mantidqtpython as mpy
import numpy

ws = simpleapi.WorkspaceFactory.create("Workspace2D", 1000, 2,1)
simpleapi.AnalysisDataService.Instance().addOrReplace('POWDIFF_test', ws)
simpleapi.LoadInstrument(Workspace=ws, Filename='data/POWDIFF_Definition.xml', RewriteSpectraMap=True)
ws = simpleapi.Rebin(ws, "0,0.1,2")
simpleapi.AnalysisDataService.Instance().addOrReplace('POWDIFF_test', ws)
InstrumentWidget = mpy.MantidQt.MantidWidgets.InstrumentWidget
app = PyQt4.QtGui.QApplication(sys.argv)
iw = InstrumentWidget("POWDIFF_test")
iw.show()

x = 1
def addtoY():
    x =1
    while 1:
    	time.sleep(4)
    	x+=1
	print x
    	ws.dataY(1)[9] += 1
	ws.dataY(1)[8] +=2
	ws.dataY(2)[1]+=1
	print ws

addThread = Thread(target=addtoY)
addThread.start()
app.exec_() 
```

LamarMoore, 2016-04-01:
Hi Lottie, I added a line to your code which triggers the workspace observer that the workspace has been modified. This may not be entirely correct, I am discussing with Owen.
```python
from threading import Thread
import time
import sys
import PyQt4
import mantid.simpleapi as simpleapi
import mantidqtpython as mpy
import numpy

ws = simpleapi.WorkspaceFactory.create("Workspace2D", 1000, 2,1)
simpleapi.AnalysisDataService.Instance().addOrReplace('POWDIFF_test', ws)
simpleapi.LoadInstrument(Workspace=ws, Filename='data/POWDIFF_Definition.xml', RewriteSpectraMap=True)
ws = simpleapi.Rebin(ws, "0,0.1,2")
simpleapi.AnalysisDataService.Instance().addOrReplace('POWDIFF_test', ws)
InstrumentWidget = mpy.MantidQt.MantidWidgets.InstrumentWidget
app = PyQt4.QtGui.QApplication(sys.argv)
iw = InstrumentWidget("POWDIFF_test")
iw.show()

x = 1
def addtoY():
    x =1
    while 1:
        time.sleep(4)
        x+=1
        print x
        ws.dataY(1)[9] += 1
        ws.dataY(1)[8] +=2
        ws.dataY(2)[1]+=1
        simpleapi.AnalysisDataService.Instance().addOrReplace('POWDIFF_test', ws) #added this line
        print ws
    
addThread = Thread(target=addtoY)
addThread.start()
app.exec_() 
```

PR #15789 (2016-05-05): Instrument View colour scale now updating
Modified instrument widget update code to include colour map and instrument view.

**To test:**

Run the below script with a developer branch version of mantidpython. 
The POWDIFF_Definition.xml is located here: https://github.com/DMSC-Instrument-Data/live-data-prototype/tree/master/data

Release notes do not need to be updated.

```python
from threading import Thread
import time
import sys
import PyQt4
import mantid.simpleapi as simpleapi
import mantidqtpython as mpy
import numpy

ws = simpleapi.WorkspaceFactory.create("Workspace2D", 1000, 2,1)
simple [... cut]
closed 2016-05-05 by AndreiSavici