mantidproject/mantid #15783: Vesuvio - VesuvioCorrections and VesuvioCommands tests failing on some platforms

ElliotAOram, 2016-03-30:
### Expected behavior
* Vesuvio Corrections and VesuvioCommands test should pass on all platforms
### Actual behavior
* Failure on some platforms
### Platforms affected
* VesuvioCommands failing on:
  * OSX
  * Ubuntu
  * RHEL6
* VesuvioCorrections failing on:
  * Fedora

ElliotAOram, 2016-03-30:
In order to avoid stopping the nightly build a temporary fix of stopping the tests running on failing OSs has been created #15784

FedeMPouzols, 2016-03-30:
On my debian machine VesuvioCommands passes.

VesuvioCorrestions fails:
```
532: ======================================================================
532: FAIL: test_gamma_and_ms_correct_workspace_index_one (__main__.VesuvioCorrectionsTest)
532: ----------------------------------------------------------------------
532: Traceback (most recent call last):
532:   File "/home/fedemp/mantid-repos/mantid/Framework/PythonInterface/test/python/plugins/algorithms/VesuvioCorrectionsTest.py", line 73, in test_gamma_and_ms_correct_workspace_index_one
532:     self._validate_matrix_peak_height(corrections_wsg.getItem(0), corrections_gb_peak)
532:   File "/home/fedemp/mantid-repos/mantid/Framework/PythonInterface/test/python/plugins/algorithms/VesuvioCorrectionsTest.py", line 379, in _validate_matrix_peak_height
532:     self.assertTrue(abs_difference <= abs(tolerance_value))
532: AssertionError: False is not true
532: 
532: ======================================================================
532: FAIL: test_gamma_and_ms_correct_workspace_index_two (__main__.VesuvioCorrectionsTest)
532: ----------------------------------------------------------------------
532: Traceback (most recent call last):
532:   File "/home/fedemp/mantid-repos/mantid/Framework/PythonInterface/test/python/plugins/algorithms/VesuvioCorrectionsTest.py", line 131, in test_gamma_and_ms_correct_workspace_index_two
532:     self._validate_matrix_peak_height(corrections_wsg.getItem(0), corrections_gb_peak)
532:   File "/home/fedemp/mantid-repos/mantid/Framework/PythonInterface/test/python/plugins/algorithms/VesuvioCorrectionsTest.py", line 379, in _validate_matrix_peak_height
532:     self.assertTrue(abs_difference <= abs(tolerance_value))
532: AssertionError: False is not true
```
I inserted a print in `_validate_matrix_peak_height()` and I got this:
```
expected_height: 6.170476e-07, peak_height: 4.22729679177e-07
abs_difference: 1.94317920823e-07, tolerance_value: 3.085238e-08
```
and this:
```
expected_height: 1.605327e-06, peak_height: 1.22509515776e-06
abs_difference: 3.80231842239e-07, tolerance_value: 8.026635e-08
```

ElliotAOram, 2016-03-30:
Thanks @FedeMPouzols I'll address this in #15784

PR #15948 (2016-04-22): Vesuvio - VesuvioCorrections and VesuvioCommands test should now pass on all platforms
* VesuvioCorrectionsTest Input parameters have been updated to improve the fit for the data
* Guard for older versions of boost has been added (was producing incorrect behaviour with random numbers #15937)

**To test:**
* Ensure the system tests pass on the build servers
* Run the system tests locally on a variety of OSs to ensure that the change works for various OSs

Fixes #15783 .

*Does not need to be in the release notes.*

---

#### Reviewer ####

Please comment on the following ([full description](htt [... cut]
closed 2016-04-22 by peterfpeterson