mantidproject/mantid #15812: Avoid Reload of Data File in ISIS SANS Reduction

picatess, 2016-04-04:
__Issue__

The data files in ISIS SANS Reductions are reloaded at least twice for each reduction. This can make sense when the data file has been changed during or just before the reduction process. But it also creates an inefficiency, since we reload the data file even if it has not been changed.

Note that this is part of a `RoadMap` issue.

__ Suggested Solution__

For each data file (for both Sample and Can) create a `MantidDataFileInfoObject` which hold information regarding:
* file name
* history
* number of histograms
* size

When a certain data file is requested for loading we check first the `MantidDataFileInfoObject`s if the file is already loaded and if the has been altered. If this is not the the case, we avoid reloading the file.

AntonPiccardoSelg, 2016-05-13:
Started to work on this. Avoiding to reload seemed quite feezible in the current reduction system and progress was made.

A meeting with the SANS Scietists concluded though that this should be put on hold and we should start with a rewrite of the ISIS SANS reduction system

AntonPiccardoSelg, 2017-10-05:
We now have a new reduction backend which does this

closed 2017-10-05 by AntonPiccardoSelg