mantidproject/mantid #15843: Python algorithm to calculate the phonon density of states from powder inelastic data

mducle, 2016-04-06:
It would be useful to have an algorithm to compute the (neutron weighted) density of states from S(q,w) in the incoherent approximation (from e.g. Skold and Price, Neutron Scattering, eqn. 1.70 http://bit.ly/1PTuc6a )

Z(E) \propto S(Q,E) / (Q^2/E) / exp(-2W) / <n+1>

for neutron energy loss; <n+1> should be <n> for neutron energy gain, with <n+1> = 1/(exp[E/kT]-1)+1, and <n> = 1/(exp[E/kT]).

This would be useful in lattice dynamics measurements in powder / polycrystalline samples, e.g. on Mari.

mducle, 2016-04-07:
I've got an algorithm written, but just need to sort out the unit conversion to get the DOS into the correct units, assuming the data is in absolute units... It needs they user to have already used SetSampleMaterial().

mducle, 2016-04-07:
Nick, is it ok if I modify SetSampleMaterial() to also set the relative molecular mass (rmm)? Or would it be ok to expose getAtom() to Python? Essentially, I need to multiply a processed and reduced inelastic dataset by the total scattering cross-section and divide by the relative molecular mass to get the density of states in the correct units. The cross-section is computed by SetSampleMaterial() if the user specifies the chemical formula, but the molecular mass is not, even though getAtom() has a list of all the elemental masses.

SpencerHowells, 2016-04-08:
I agree with this. I asked for something similar a very long time ago.

SpencerHowells, 2016-04-08:
Further to the previous comment, absorption corrections need n*sigma where n is the number density & sigma cross-sections. We have have always used these parameters as average per atom. In SetSampleMaterial it would be good to have as output the cross-sections per atom and the average mass calculated. if the density (specific gravity) is input it could also calculate n. This would make it much easier for the users (and less prone to mistakes!) as they would only need to input the formula and the measured density (g/cc).

mducle, 2016-04-08:
@SpencerHowells I'm not sure I understand what you mean in your first comment - are you referring to  the exposing the C++ function "getAtom" to Python; modifying the Materials class to include the relative molecular mass; or about computing the incoherent phonon DOS? 

Regarding your second comment, I understand that you would like SetSampleMaterial to calculate the number density if the user gives the chemical formula and the measured density in (g/cc). Is this right? I which case, we should probably set up a separate issue for that, and I can make a separate pull request.

SpencerHowells, 2016-04-11:
1. Inelastic spectrometers also measure S(q,w) and I was just about to start writing an algorithm to compute the DOS! I look forward to seeing your version and I will test it on our IRIS/OSIRIS data.
2. I have written corrections programs for the Inelastic package and use SetSampleMaterial  a lot but it does not do a few things that I need, such as molecular mass and the average atomic mass. I thought that exposing getAtom might help but on reflection I don't think it does.
3. Calculating the number density is, as you say, probably a separate Issue.

NickDraper, 2016-04-11:
Absolutely go ahead and make those changes, if you need any help about exposing objects to Python, drop by the office and talk to Martyn Gigg

mducle, 2016-04-20:
@SpencerHowells I've made a pull request for this. Is there some IRIS or OSIRIS datafiles I can test it on?

closed 2016-05-05 by FedeMPouzols