mantidproject/mantid #15856: Performance test LoadEventNexusTestPerformance affected by StructuredDetector

OwenArnold, 2016-04-07:
Looking at the performance test, I can see a significant jump in the execution time for these tests following the introduction of #15744 

[image]

### Steps to reproduce the behaviour

You should run the performance tests yourself before and after merge commit dc9168cc48abd3f5689c8371c0d7949dba674438 to see the timings for yourself and then identify and fix the issue.

### Platforms affected

Likely all platforms

OwenArnold, 2016-04-07:
@mantid-roman, @LamarMoore  is away for a few weeks, do you have time to investigate this before his return?

OwenArnold, 2016-04-07:
This particular performance test might be pointing more towards the root cause of the problem.
[image]

OwenArnold, 2016-04-18:
@MikeHart85 Could you please build between these change-sets on Ubuntu and see if you get the 2* speed change the performance tests indicate. If so, could you run callgrind and determine where the slow down might be so that Lamar can fix it.

MikeHart85, 2016-05-04:
I had the following results (in secs) when comparing commits `12b9eb521167b99810c127f3d688794867693e47` (before) and `dc9168cc48abd3f5689c8371c0d7949dba674438` (after):

Test | Class | Function | Before | After
----- | -------- | ----------- | --------- | ------
DataHandlingTest | LoadEventNexusTestPerformance | testDefaultLoad | 0.67 | 1.06
DataHandlingTest | LoadEventNexusTestPerformance | testDefaultLoadBankSplitting | 0.73 | 0.36
DataHandlingTest | LoadEventPreNexusTestPerformance | testDefaultLoad | 0.25 | 0.90
GeometryTest | QuadrilateralTestPerformance | test_Area_Calls | 3.92 | 6.18

Note the decrease on the BankSplitting test. Not sure why it's so much faster, the profiler showed it using about 40% more cycles oddly enough.

Quick look over the Valgrind output with Lamar showed that the most likely cause of the slowdown on other tests was the introduction of regular expressions in the InstrumentDefinitionParser and the Detector classes.

quantumsteve, 2016-05-04:
I don't see any regular expressions or recent changes in `test_Area_Calls` or the functions it calls. Any ideas why that might be so much slower?

dreitz, 2016-05-04:
ZzzsszzzssZsssss szszzssszzzzzzzzzzz DS sszszzssszzzzdzzZse Z
szszzzzzzzZzzzzzzzzzzzzZszzzzszzZzZszzzzzzzzzzzzzZzzzzzzzzzESDSss4zrzszssssZ
dzsis D D dss ssZSseS sZZse ek SDS DS DS SSS sssss se sssss zdszzzzszz s
assuredSz sqes sseeness3sd2zSD sdss sszSrrzZzsSsSerrorS sssszzzzzzzzzzsssss

MikeHart85, 2016-05-05:
No, that one was the most confusing result. Most of the time is reported as being spent in malloc.c and stl_deque.h. Between the two commits, it looks like memory allocation got slightly faster but deallocation got much slower. But I couldn't see any way the changes from this PR could be involved.

My plan right now is to revert the regex changes and then rerun the tests. Perhaps the results will give a hint as to what is going on in there.

MikeHart85, 2016-05-05:
This is a table comparing previous before and after timings against timings with the regex changes reverted. So using dc9168c as a starting point and just replacing the introduced regex matches with string compares. Omitting Test and Class columns, see table in previous comment for those.

Function | Before | After | After, No Regex
----------- | --------- | ------ | --------------------
testDefaultLoad | 0.67 | 1.06 | 0.44
testDefaultLoadBankSplitting | 0.73 | 0.36 | 0.29
testDefaultLoad | 0.25 | 0.90 | 0.27
test_Area_Calls | 3.92 | 6.18 | 6.15

Ran all tests multiple times to confirm, those are fairly stable average values. `test_Area_Calls` fluctuates between 6.13 and 6.16. The others even less, varying at most by 0.01.

Both `LoadEventNexusTestPerformance` tests show significant improvement over the original, from before dc9168c. I assume the regex slowdown was outweighing performance improvements elsewhere. `LoadEventPreNexusTestPerformance` seems back to normal with negligible difference. 

`QuadrilateralTestPerformance` is still significantly slower than before. The regex code didn't seem to affect it at all. Which makes sense. But that means there's still something else in this PR that slowed it down.

MikeHart85, 2016-05-05:
One thing that stands out is that, in both "After" builds, valgrind is showing a chain of calls to `free` -> `_int_free` -> `malloc_consolidate` as part of deleting the `std::deque`s of `Kernel::V2D`s in `ConvexPolygon`.

The "Before" build only calls `free` -> `_int_free` and doesn't invoke `malloc_consolidate` at all.

All of these are invoked by implicit deletion as things are going out of scope. Deletion is invoked 100M times, consolidation 50M - 1 times.

Haven't been able to find any connection to Lamar's changes so far. There's not much code involved in this QuadrilateralTest to begin with, it seems quite isolated. Starting to suspect this is just the optimizer and/or std library behaving differently because things have shifted around in memory.

quantumsteve, 2016-05-05:
Do you have access to OS X or Windows? On my 2012 Mac Pro, GeometryTest_QuadrilateralTestPerformance only takes 1.25 seconds.

MikeHart85, 2016-05-05:
I have a dual boot setup with Windows, but I usually work on Linux so my Windows setup is horribly out of date.

But Lamar ran the Quadrilateral test on his Windows setup and didn't see any difference. Both before and after these changes it took about 4 seconds for him.

Valgrind shows most of the time in this test being spent in construction and deletion of the V2D deques. So it's likely that performance depends mostly on memory speed and quality of the memory manager library.

Seems like on Ubuntu 14.04, eglibc-2.19 is making some poor choices after the changes in this PR are added... though I don't know how Lamar's changes would cause this.

@OwenArnold mentioned it might be worthwhile to leave this as is for this issue (just replace regex matches with string comparisons) and open another for the Quarilateral test being slow on Linux.

OwenArnold, 2016-05-05:
I don't like any of the "fixes" we could apply to get around the `malloc_consolidate` issue particularly when it comes to hand-rolling our own allocation/de-allocation. As @MikeHart85 says, we should treat this as a separate issue if we can handle it at all.

quantumsteve, 2016-05-05:
@OwenArnold The issue goes away if we replace std::deque with std::vector in ConvexPolygon. I don't see any benefits to using a deque and will submit a pull request with this change.

PR #16083 (2016-05-11): Fix performance regressions caused by regex matches 
This addresses #15856 by optimizing the regex matches introduced in #15744.

Originally, regex matches introduced a significant slowdown. 

Reverting to string comparisons was explored as an option, but ultimately it turned out that the regex code could easily be optimized to nearly the same performance levels. 

Just making the regex expression definition `static` had a profound effect. Should be considered for other usages as well, where the expression is constant.

**To test:**

Run `DataHandlingTest LoadEventNexusTestPerformance` and `Da [... cut]
closed 2016-05-11 by OwenArnold