mantidproject/mantid #15857: Buffer overruns in Geometry tests, ObjCompAssembly

FedeMPouzols, 2016-04-07:
### Expected behavior
Unit tests should pass

### Actual behavior
These two tests fail on win7 / debug mode:
- `GeometryTest_ObjCompAssemblyTest`
- `GeometryTest_ParObjCompAssemblyTest` 

The problem seems to be in `GluGeometryHandler::GetObjectGeom` with `SEGMENTED_CYLINDER`. `vector` has size 0
This is a stack trace for `ParObjCompAssemblyTest`:
```
 	msvcp140d.dll!std::_Debug_message(const wchar_t * message, const wchar_t * file, unsigned int line) Line 17	C++
 	msvcp140d.dll!std::_Debug_message(const unsigned short * message, const unsigned short * file, unsigned int line) Line 23	C++
 	MantidGeometry.dll!std::vector<Mantid::Kernel::V3D,std::allocator<Mantid::Kernel::V3D> >::operator[](unsigned __int64 _Pos) Line 1233	C++
>	MantidGeometry.dll!Mantid::Geometry::GluGeometryHandler::GetObjectGeom(int & mytype, std::vector<Mantid::Kernel::V3D,std::allocator<Mantid::Kernel::V3D> > & vectors, double & myradius, double & myheight) Line 142	C++
 	MantidGeometry.dll!Mantid::Geometry::Object::GetObjectGeom(int & type, std::vector<Mantid::Kernel::V3D,std::allocator<Mantid::Kernel::V3D> > & vectors, double & myradius, double & myheight) Line 1963	C++
 	GeometryTest.exe!ParObjCompAssemblyTest::testCreateOutlineCylinder() Line 141	C++
 	GeometryTest.exe!TestDescription_suite_ParObjCompAssemblyTest_testCreateOutlineCylinder::runTest() Line 57	C++
 	GeometryTest.exe!CxxTest::RealTestDescription::run() Line 99	C++
 	GeometryTest.exe!CxxTest::TestRunner::runTest(CxxTest::TestDescription & td) Line 95	C++
 	GeometryTest.exe!CxxTest::TestRunner::runSuite(CxxTest::SuiteDescription & sd) Line 81	C++
 	GeometryTest.exe!CxxTest::TestRunner::runWorld() Line 64	C++
 	GeometryTest.exe!CxxTest::TestRunner::runAllTests(CxxTest::TestListener & listener) Line 26	C++
 	GeometryTest.exe!CxxTest::XUnitPrinter::run() Line 31	C++
 	GeometryTest.exe!CxxTest::Main<CxxTest::XUnitPrinter>(CxxTest::XUnitPrinter & tmp, int argc, char * * argv) Line 107	C++
 	GeometryTest.exe!main(int argc, char * * argv) Line 27	C++
 	[External Code]	
```
And this is the call stack for the failure in `ObjCompAssemblyTest`:
```
 	msvcp140d.dll!std::_Debug_message(const wchar_t * message, const wchar_t * file, unsigned int line) Line 17	C++
 	msvcp140d.dll!std::_Debug_message(const unsigned short * message, const unsigned short * file, unsigned int line) Line 23	C++
 	MantidGeometry.dll!std::vector<Mantid::Kernel::V3D,std::allocator<Mantid::Kernel::V3D> >::operator[](unsigned __int64 _Pos) Line 1233	C++
>	MantidGeometry.dll!Mantid::Geometry::GluGeometryHandler::GetObjectGeom(int & mytype, std::vector<Mantid::Kernel::V3D,std::allocator<Mantid::Kernel::V3D> > & vectors, double & myradius, double & myheight) Line 142	C++
 	MantidGeometry.dll!Mantid::Geometry::Object::GetObjectGeom(int & type, std::vector<Mantid::Kernel::V3D,std::allocator<Mantid::Kernel::V3D> > & vectors, double & myradius, double & myheight) Line 1963	C++
 	GeometryTest.exe!ObjCompAssemblyTest::testCreateOutlineCylinder() Line 366	C++
 	GeometryTest.exe!TestDescription_suite_ObjCompAssemblyTest_testCreateOutlineCylinder::runTest() Line 129	C++
 	GeometryTest.exe!CxxTest::RealTestDescription::run() Line 99	C++
 	GeometryTest.exe!CxxTest::TestRunner::runTest(CxxTest::TestDescription & td) Line 95	C++
 	GeometryTest.exe!CxxTest::TestRunner::runSuite(CxxTest::SuiteDescription & sd) Line 81	C++
 	GeometryTest.exe!CxxTest::TestRunner::runWorld() Line 64	C++
 	GeometryTest.exe!CxxTest::TestRunner::runAllTests(CxxTest::TestListener & listener) Line 26	C++
 	GeometryTest.exe!CxxTest::XUnitPrinter::run() Line 31	C++
 	GeometryTest.exe!CxxTest::Main<CxxTest::XUnitPrinter>(CxxTest::XUnitPrinter & tmp, int argc, char * * argv) Line 107	C++
 	GeometryTest.exe!main(int argc, char * * argv) Line 27	C++
 	[External Code]	
```

### Steps to reproduce the behavior

Run the tests in debug mode.

### Platforms affected

Windows 7

FedeMPouzols, 2016-04-07:
@LamarMoore , @OwenArnold this may be related to recent changes in geometry? The two failures have happened for the last three nights at least.

quantumsteve, 2016-04-07:
This is fixed in PR #15842.

FedeMPouzols, 2016-04-11:
Closing this as fixed in PR #15842.

closed 2016-04-11 by FedeMPouzols