mantidproject/mantid #15864: Generalise LoadReflTBL/SaveReflTBL

raquelalvarezbanos, 2016-04-08:
`LoadReflTBL` and `SaveReflTBL` should be generalized as part of the work on the data processor user interface. At present, they are specific to Reflectometry in terms of the number of columns (which is fixed, and equal to 9) and the type of each column.

MattKing06, 2016-04-20:
There are currently some obstacles stopping this from being implemented as smoothly as we had planned.

It seems that when the CSV values are read from a table they contain only the values for that row with no indication of what column they are to be assigned to. In our current code for LoadReflTBL we are predefining the column headings and types like this:

```
  auto colRuns = ws->addColumn("str", "Run(s)");
  auto colTheta = ws->addColumn("str", "ThetaIn");
  auto colTrans = ws->addColumn("str", "TransRun(s)");
  auto colQmin = ws->addColumn("str", "Qmin");
  auto colQmax = ws->addColumn("str", "Qmax");
  auto colDqq = ws->addColumn("str", "dq/q");
  auto colScale = ws->addColumn("double", "Scale");
  auto colStitch = ws->addColumn("int", "StitchGroup");
  auto colOptions = ws->addColumn("str", "Options");

  colRuns->setPlotType(0);
  colTheta->setPlotType(0);
  colTrans->setPlotType(0);
  colQmin->setPlotType(0);
  colQmax->setPlotType(0);
  colDqq->setPlotType(0);
  colScale->setPlotType(0);
  colStitch->setPlotType(0);
  colOptions->setPlotType(0);
```

Once we have split the comma-separated line into individual values, we then assign each value to it's predefined column i.e

"14456, 0.5, 1.000494, 2.9494, 0.1" is split into vector which would read like this `vector[0] = 14456`
`vector[1] = 0.5` etc. Once we have this vector, it is then easy to map the value to it's column heading as the value at the first index would go to the first column and so forth.

The difficulty we run into when trying to make this generic is that we should not be allowed to predefine the column headings as each scientific area may require different column headings.

_Proposed solution:_

- Make the first line of any .TBL file contain a comma-separated string of column headings such as
 "Runs,Angle,QMin,QMax"
- Read the number of entries in this line and that will give us the number of columns.
- Use each value from the string to set the heading of that column

_issues with proposed solution_
- This can affect the backward compatibility as any current .TBL does not contain this column headings line and therefore could not be loaded properly into the interface using this algorithm

_fix for these issues_
- If a CSV file does not contain this column heading line, then we will resort back to using the predefined list of column headings. This will hopefully work as the only technique area to use these algorithms and TBL files is reflectometry and the predefined column headings should suffice.

MattKing06, 2016-06-14:
@raquelalvarezbanos Can this issue be closed now?

raquelalvarezbanos, 2016-06-14:
Yes, thanks @MattKing06, it can be closed now. Please use label "Review required" and explain why it can be closed and why it was not automatically closed when the corresponding PR was closed.

MattKing06, 2016-06-14:
@mantidproject/gatekeepers The fix for this issue can be found in #15968 as can the reasons for why this issue was not closed when the PR was closed.

closed 2016-10-03 by NickDraper