mantidproject/mantid #15868: PredictPeaks on LET

OwenArnold, 2016-04-08:
PredictPeaks is failing to create allowed peaks on a LET dataset.

**How to reproduce**

From the archive you need: `/archive/ndxlet/Instrument/data/cycle_15_1/LET00021021.raw`
Use this to create `LET21021.mat` save the contents to the file system:
```
 0.13107693  0.01932425  0.00240078 
-0.00054492 -0.00061595  0.03165928 
 0.03555081 -0.11679514 -0.00118727 
     7.6307     31.5760      8.2823     90.2395     98.5314     89.8656   1973.4955 
     0.0336      0.4152      0.0286      0.7794      0.3010      0.7950     28.2070 

The above matrix is the Transpose of the UB Matrix. The UB matrix maps the column
vector (h,k,l ) to the column vector (q'x,q'y,q'z).
|Q'|=1/dspacing and its coordinates are a right-hand coordinate system where
 x is the beam direction and z is vertically upward.(IPNS convention)
```
Then run this:
```python
Load(Filename='LET00021021.raw', OutputWorkspace='LET00021021')
LoadIsawUB(InputWorkspace='LET00021021', Filename='LET21021.mat')
PredictPeaks(InputWorkspace='LET00021021', WavelengthMin=0.5, WavelengthMax=6, MinDSpacing=0.5, MaxDSpacing=36, OutputWorkspace='peaks')
```
Use the instrument view to put the predicted peak onto of the data and you get this:
[image]

There are missing nuclear reflections such as (0,0,1), (0,-1,2), (0,-1,1).

Now repeat the steps above, but instead of using Load, use LoadEmptyInstrument to load the LET IDF directly. Then you get this:

[image]

OwenArnold, 2016-04-08:
**Initial Investigations**

This does not appear to be a result of wavelength dependence. After introducing some diagnostic code to `findDetectors` in `Peak.cpp`, it looks like the reflections such as (0,0,1) are being generated, but there is a failure to assign a detector to them.

The chain of calls close to the failure is:

```PredictPeaks::exec()->Peak::Peak(HKL)->InstrumentRayTracer::getDetectorResult()``` where InstrumentRayTracer::firRay() is actually called first by `Peak::Peak(HKL)`.

What we know:

1. There is a 4:1 mapping between Detectors and Grouped detectors in the Data case, that is not present in the LoadEmptyInstrument case. However DetectorGroup is only created on the fly in the Workspace, so the Instrument itself does not contain the mapping information.
1. As follows from point above, with both data sets (empty instrument and proper data) the `Instrument` passed to `InsrumentRayTracer` should be identical. Only the mapping should be different.  So why do we get different results (detector not found) from the outputs of the ray tracer?
1. Introducing `tube-gap` into the parameter for the Data via `SetInstrumentParameter` and setting the value to 0.000014, does not resolve the issue, so its unlikely to be just the tube-gaps that are causing this.
1. The issue could possibly result from the detectors being physically located in a different position between the two workspaces. Apparently on LET it is routine to shift the detector positions, however loading from the raw file does not revel what pre-processing might have occurred, and we cannot compare detector positions directly between the two workspaces because the mapping is getting in the way.

OwenArnold, 2016-04-08:
@MichaelWedel @martyngigg FYI

martyngigg, 2016-04-08:
@OwenArnold Can you generate mapping file by hand that you can pass to `GroupDetectors` to compare the positions?

OwenArnold, 2016-04-08:
@martyngigg Is it possible to extract the mapping file from the real workspace?

martyngigg, 2016-04-08:
You can't extract it directly but `GroupDetectors` can accept another as a template to copy from.

closed 2021-03-03 by stale[bot]