mantidproject/mantid #15875: Refactor SANS back end

NickDraper, 2016-04-11:
The current SANS back end is based on the old reducer framework that is no longer used within mantid, and makes every addition to Mantid harder.

This task is to refactor the back end processes of the SANS reduction to use workflow algorithms.

Requirements:
1. This should create a few workflow algorithms  (that may call others) to perform the ISIS SANS data reduction.
1. It should share as much as possible (or be the same as) the ANSTO codes. 
1. This should be done separately from the current GUI and back end, which will remain in place until we are ready to replace it.
1. Batch processing should be considered first, and single run processing only as a special case of batch processing (this affects the GUI more than the script).

AntonPiccardoSelg, 2017-04-12:
The work on this is complete. The system is currently in the verification process and being rolled out into master. So far we have rolled out:

* [State](https://github.com/mantidproject/mantid/pull/18157)
* [Moving, masking and loading](https://github.com/mantidproject/mantid/pull/19225)
* [Slicing, QConvert, WavlengthConvert, scaling](https://github.com/mantidproject/mantid/pull/19333)
* [Adjustment workspaces](https://github.com/mantidproject/mantid/pull/19351)

To be rolled out
* Batch algorithm
* ISISCommandInterface
* Systemtests

AntonPiccardoSelg, 2017-05-25:
This issue will be released in 3.10

closed 2017-06-01 by NickDraper